fivdi/onoff #31: I/O errors on first launch

joraff, 2015-04-09:
The first time I launch a node application (as sudo) that uses onoff.js, I get this error:
```
fs.js:540
  return binding.write(fd, buffer, offset, length, position);
                 ^
Error: EIO, i/o error
    at Object.fs.writeSync (fs.js:540:18)
    at Object.fs.writeFileSync (fs.js:987:21)
    at new Gpio (/home/pi/nodeplay/node_modules/onoff/onoff.js:103:10)
    at Object.<anonymous> (/home/pi/nodeplay/process.js:15:16)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
```
Subsequent launches are fine, until the next reboot.
Any ideas?

fivdi, 2015-04-10:
Is the following the case:

* The program is creating a `Gpio` object where the direction is `"output"` and the `edge` is `"rising"`, `"falling"`, or `"both"`. 
* Linux kernel > 3.13 is being used.

`dmesg` may also be displaying messages similar to these:

```
[ 1180.184658] GPIO chip pinctrl-bcm2835: gpio_lock_as_irq: tried to flag a GPIO set as output for IRQ
[ 1180.184688] gpio-17 (sysfs): failed to flag the GPIO for IRQ
```

If this is the case, then the error is explained by the fact that the Linux 3.13 kernel dropped support for detecting changes on GPIO output pins per hardware interrupt. It can still detect interrupts on input pins.

This is documented in the onoff readme [here](https://github.com/fivdi/onoff#gpiogpio-direction-edge) :).

See also [https://github.com/raspberrypi/linux/issues/808](https://github.com/raspberrypi/linux/issues/808) and [https://github.com/torvalds/linux/commit/d468bf9ecaabd3bf3a6134e5a369ced82b1d1ca1](https://github.com/torvalds/linux/commit/d468bf9ecaabd3bf3a6134e5a369ced82b1d1ca1)

fivdi, 2015-04-10:
I think this issue is related to https://github.com/fivdi/onoff/issues/31

joraff, 2015-04-13:
That's precisely it: ```new Gpio(4, 'out', 'falling');``` What a bummer. Would a acceptable workaround be to initialize as an input, then switch to an output if I need to drive it?

fivdi, 2015-04-13:
Given that I have no information about the circuit you're building, I don't know. If everything is electrically ok, the setEdge and setDirection methods could probably be used to handle the software side of things.

joraff, 2015-04-13:
Electrically everything is fine - it was working on a RPi model B+ before I upgraded to RPi 2. I was most likely using an older version of raspbian.

closed 2015-04-13 by joraff