fivdi/onoff #33: Question about read right

MakiseKurisu, 2015-06-09:
I'm using RPi 2 with onoff. On my machine I cannot read gpio no matter I export the pin using gpio-admin or not. However as soon as I use python to read the same pin, node.js can get the value even if I unexport the pin with gpio-admin. How can this happen?

The code I'm using:
```javascript
var gpio = require('onoff').Gpio,
    http = require('http'),
    button = new gpio(18, 'in', 'both');

function exit() {
  process.stdout.write('\n');
  button.unexport();
  process.exit();
}
process.on('SIGINT', exit);

button.watch(function(err, state) {
  if (err) {
    throw err;
  }

  process.stdout.write(button.readSync() + '');
});
 
http.createServer(function(req, res) {
  res.end(button.readSync() + '');
  process.stdout.write(button.readSync() + '');
}).listen(8080);
```

```python
import RPi.GPIO as GPIO
import time

print('Setup GPIO pin 18')
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

print('Setup LED')
f = open('/sys/class/leds/led1/trigger', 'r+')
f.write('gpio\n')
f.close()

print('Ready')

while True:
    input_state = GPIO.input(18)
    if input_state == False:
        f = open('/sys/class/leds/led1/brightness', 'r+')
        if f.read() == '1\n':
            f.write('0\n')
        else:
            f.write('1\n')
        f.close()
        print('Button Pressed')
        time.sleep(0.2)
```

fivdi, 2015-06-09:
To answer this question correctly it would be necessary to know how the button is connected to the RPi 2, so I'm going to have to guess here.

Here's the guess:

GPIO 18 is being used as the input and the pull-down resistor for GPIO 18 is activated. The pull-down resistor is activated by default for GPIO 18 as can be seen in table 6-31 on page 102 of the [BCM2835 ARM Peripherals documentation](http://www.farnell.com/datasheets/1521578.pdf). Because the pull-down resistor is activated, the value of GPIO 18 is 0 when the button is not pressed. When the button is pressed, it's pulled to 0V, so there is no change in it's value and no change can be detected.

In the Python program the pull-up resistor is activated, so it functions as expected.

If this is the case, using a GPIO which is pulled-up by default (see table 6-31) should solve the problem. Alternatively, the pull-up for GPIO 18 could be activated as described [here](https://github.com/fivdi/onoff/wiki/Enabling-Pullup-and-Pulldown-Resistors-on-The-Raspberry-Pi)

Note that the [gpio-admin](https://github.com/quick2wire/quick2wire-gpio-admin) has been abandoned, and I wouldn't recommend using it any more. It does not function correctly on the RPi 2, see [here](https://github.com/quick2wire/quick2wire-gpio-admin/issues/5).

fivdi, 2015-06-11:
@MakiseKurisu is this still an issue?

MakiseKurisu, 2015-06-13:
Thank you. That is the issue. I'm using pin 5 and everything is perfectly fine.

I installed gpio-admin since the guide in adafruit has it. Maybe you want to add a warning in your readme.md.

fivdi, 2015-06-13:
Ok, that's good to hear and thanks for the feedback.

The [Adafruit Learning System](https://github.com/fivdi/onoff#adafruit-learning-system) section of the readme has been updated to warn against using the gpio-admin tool. I also sent a message to Adafruit and hopefully they'll update the article.

closed 2015-06-13 by fivdi