fivdi/onoff #35: Always returning 1 as new value during toggling

roccomuso, 2015-06-29:
Someone can explain why it returns always 1 as new value? It can't toogle.

	var Gpio = require("onoff").Gpio;
	led = new Gpio(17, 'out');

	   led.read(function (err, value) { // Asynchronous read.
			if (err) throw err;

			led.write(value ^ 1, function (err) { // Asynchronous write.
			  if (err) throw err;
			  console.log('Pin new value: '+stato);
			});
	  });

fivdi, 2015-06-30:
Is it because `stato` is being logged to the console instead of `value`?

roccomuso, 2015-07-01:
No it's just a typo in the question. Fixed. 

PS. I'm using Raspbian on a Raspberry Pi B.

fivdi, 2015-07-02:
I would expect `stato` to have a value of `off` in the example program, so everything appears to be function correctly.

This line of code creates a new Gpio object called `led` with an initial value of 0:
```
led = new Gpio(17, 'out');
```
This line of code reads the value of `led` and sets `value` to 0:
```
led.read(function (err, value) { // Asynchronous read.
```
This line of code therefore sets `stato` to `off`
```
var stato = (value == 1) ? "on": "off";
```

What makes you think `salto` should be equal to `on`?

PS. Please don't modify previous posts as it makes it difficult/impossible to understand the topic later.

roccomuso, 2015-07-02:
So how can i read the value of led without setting it to 0?

Doesn't exist a way to read the pin state?

And for example how would you implement a toggling function without reading
the current pin value?

fivdi, 2015-07-02:
There appears to be a misunderstanding here. Reading the value of a Gpio object for a LED will ***NOT*** set it's value to 0. What will set it's value to 0 is creating a new Gpio object for the LED. If the program is continuously creating Gpio objects for the LED, then the value will continuously be set to 0. This should be avoided.

Please try the following program and let us know if it functions correctly. It toggles the state of a LED on GPIO 17 every 200ms, 25 times:
```js
var Gpio = require('onoff').Gpio,
  led = new Gpio(17, 'out');

(function blink(count) {
  if (count <= 0) {
    return led.unexport();
  }

  led.read(function (err, value) {
    led.write(value ^ 1, function (err) {
    });
  });

  setTimeout(function () {
    blink(count - 1);
  }, 200);
}(25));
```

roccomuso, 2015-07-02:
As you suggested, every new Gpio object sets the value to 0. Thanks, now all works fine ;)

fivdi, 2015-07-03:
That's good news :)

closed 2015-07-03 by fivdi