fivdi/onoff #36: Register pin without altering state

domoconnell, 2015-08-07:
Great library :+1: 

Not really an issue, more like a feature request.

When a pin is registered at the start of the script using
var led = new Gpio(14, 'out');

The state of pin 14 is set to LOW (0), regardless of what it was before. If a previous script has set the pin to HIGH and then finishes, or the script restarts for whatever reason, the pin is always reset to LOW.

Is it possible to have the pin registered, but not have the start altered until it is done so manually using write or writeSync?

I appreciate your time.

fivdi, 2015-08-07:
No, this currently isn't possible. If persistent state is needed between different runs of the script, it would be better to store the required information using the file system. Also, if this feature was implemented, it would be unclear what state the output pin should have the first time the script is run. It would also be unclear what state the pin should have if the system is powered down between two runs of the script.

fivdi, 2015-08-28:
I'm going to close this one as onoff isn't the correct place for saving/restoring the state of pins between different runs of the application.

closed 2015-08-28 by fivdi