fivdi/onoff #37: node-gyp configure error: gyp_main.py

pradhanshrijal, 2015-08-28:
Error while trying: node-gyp configure on beaglebone black

[image]

fivdi, 2015-08-28:
The error
```
gyp_main.py: error: no such option: --no-parallel
```
is a node-gyp issue that has been reported several times, for example, https://github.com/nodejs/node-gyp/issues/426 and https://github.com/nodejs/node-gyp/issues/434.

One workaround for this issue that has worked for me in the past is to directly modify the node-gyp file called configure.js. For example, on your machine node-gyp v2.0.2 is installed. Removing [this line](https://github.com/nodejs/node-gyp/blob/2.0.2/lib/configure.js#L329) from configure.js should fix the `no such option: --no-parallel` error.

However, although it looks like you're using the latest recommended Debian image (2015-03-01) on the BeagleBone Black, it comes with an old version of the GNU C/C++ compiler, version 4.6.3. Unfortunately, this version of the compiler is no longer capable of compiling the epoll addon which onoff deponds on due to the following nan issue: https://github.com/nodejs/nan/issues/394#issuecomment-135793288.

fivdi, 2015-08-30:
I just tried the following on the BeagleBone Black:

1) Prepared a fresh install of the [latest recommended Debian image (2015-03-01)](http://beagleboard.org/latest-images). This version of Debian comes with node v0.10.29. Unfortunately, this version of node is broken as it doesn't carry the [REPLACE_INVALID_UTF8 patch](https://github.com/joyent/node/commit/881ac26f27f4ac9585d66c8d8a67d5b246a23d1b).

2) Installed the NodeSource distribution of node v0.12.7. Installation instructions are available [here](https://github.com/nodesource/distributions#installation-instructions).

3) Modified `/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js` and removed [this](https://github.com/nodejs/node-gyp/blob/2.0.1/lib/configure.js#L327) line. This line needs to be removed to work around the "no such option: --no-parallel" error.

4) `npm install onoff` which output the following:
```
root@beaglebone:~/dev/onoff-v0.12.7# npm install onoff
/
> epoll@0.1.15 install /root/dev/onoff-v0.12.7/node_modules/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory `/root/dev/onoff-v0.12.7/node_modules/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory `/root/dev/onoff-v0.12.7/node_modules/onoff/node_modules/epoll/build'
onoff@1.0.2 node_modules/onoff
└── epoll@0.1.15 (bindings@1.2.1, nan@2.0.8)
root@beaglebone:~/dev/onoff-v0.12.7# 
```
The installation was successful and everything looks ok.

closed 2015-08-30 by fivdi