fivdi/onoff #38: Support pull-up and pull-down naively as other libs do...

born2net, 2015-10-02:
Support pull-up and pull-down naively as other libs do...
no real info how to use the tree and really it should be transparent to user, just set flag

tx

Sean

fivdi, 2015-10-02:
Are you aware of a technique for performing this in a platform independent way?

There's info about how to use the tree on the Raspberry Pi [here](https://github.com/fivdi/onoff/wiki/Enabling-Pullup-and-Pulldown-Resistors-on-The-Raspberry-Pi) although it is far from transparent.

born2net, 2015-10-02:
yes but it would be great if we could control through the lib, just like Python does it...

fivdi, 2015-10-02:
Can the Python lib do it without sudo?

born2net, 2015-10-02:
I believe it does need sudo, but for me and many others that's just fine as I run as root, or use helper exec such as utlity runuser to for added privileges.

I think onoff is the PERFECT lib, and it's the last thing it's missing to be the BEST!

fivdi, 2015-10-02:
Thank you for your kind words, but I don't know if I want to add a C++ addon to onoff for achieving this.

If this is the last thing missing to make onoff the BEST, then what, in your opinion, is the BEST today?

born2net, 2015-10-03:
I can tell you that the most popular hardware today for the raspberry pi
are the adafruit hats, servo, lcd, motors, and they all use i2c. no one had
translated their api to node, and that would be very awesome...after u add
the pull up down resistor option ;)

fivdi, 2015-10-03:
onoff doesn't support I2C (but https://github.com/fivdi/i2c-bus does). I'm not sure I fully understand now. Is the pullup/pulldown support you need related to I2C?

born2net, 2015-10-03:
No I just needed to have the ability to change the pull up and pull down on
projects that I'm using the library for

fivdi, 2015-10-03:
It looks like it should be possible to achieve this on the Raspberry Pi without requiring the use of sudo.

born2net, 2015-10-03:
You could add that that would be awesome

fivdi, 2015-11-01:
@born2net and anyone else interested in GPIO on the Raspberry Pi.

onoff is actually a platform independent package that functions on multiple systems like the Raspberry Pi and the BeagleBone. Rather than extending onoff with Raspberry Pi specific code, I decided to implement a new package called [pigpio](https://github.com/fivdi/pigpio) that is specifically tailored to the Raspberry Pi. pigpio already supports the following [features](https://github.com/fivdi/pigpio#features):

##### Features
 * Digital read and write
 * PWM on any of gpios 0 through 31
 * Servo control on any of gpios 0 through 31 (untested)
 * Pull up/down resistors
 * Interrupt handlers

So @born2net, I'd suggest taking a look at [pigpio](https://github.com/fivdi/pigpio) and recommend using it on the Raspberry Pi when onoff doesn't fit the requirements.

born2net, 2015-11-01:
sure, will check it out tx!!!

fivdi, 2015-11-02:
@born2net I'm going to close this issue.

If Linux is ever extended to provide a platform independent mechanism for configuring pull-up and pull-down resistors, onoff will also be extended to make use of such a mechanism. onoff is very unlikely to be extended to make use of platform specific mechanisms.

The alternative in this case is to use platform specific libraries such as [pigpio](https://github.com/fivdi/pigpio).

closed 2015-11-02 by fivdi