fivdi/onoff #40: Cannot compile on Raspbian on Raspberry Pi B 2 due to REPLACE_INVALID_UTF8 error (with solution)

dustinbolton, 2015-12-01:
This may be worth noting in the documentation since it took me a long time to figure out.  The latest Raspbian comes with a faulty Node v0.10.29 which is missing a UTF8 patch.  This results in your epoll dependency failing to compile.  Note gcc is v4.9.2.

The error shown is:
```
../node_modules/nan/nan.h:328:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
```

The only solution I found is to upgrade to a newer version, eg:
```
curl -sL https://deb.nodesource.com/setup_0.12 | sudo -E bash -
sudo apt-get install -y nodejs
```

This allows onoff to install as expected.  It's unfortunate that the out of box Raspbian has this issue but hopefully others won't have to run into this issue too.

fivdi, 2015-12-01:
Hmm, that's unfortunate. Thanks for the information. Debian for the BeagleBone Black had the same issue with Node.js v0.10.29. It means that practically no native add-ons can be installed. I'll add a notice to the README.

fivdi, 2015-12-02:
This issue has been documented [here](https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi) in the wiki and wiki page has been linked to in the readme.

@dustinbolton thanks again for the information.

closed 2015-12-02 by fivdi