fivdi/onoff #42: symbol lookup error

rampIc15, 2015-12-20:
I configured a GPIO ( GPIO 25 to be exact and pulled up) 
When I press the switch, the following error occurs 

```
node: symbol lookup error: /home/pi/mytools/gateway/node_modules/onoff/node_modules/epoll/build/Release/epoll.node: undefined symbol: _ZN4node12MakeCallbackEN2v86HandleINS0_6ObjectEEENS1_INS0_8FunctionEEEiPNS1_INS0_5ValueEEE
```

I tried upgrading wheezy but still did not resolve the issue. Current version I am at  : 

```
Linux version 4.1.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #817 PREEMPT Sat Sep 19 15:25:36 BST 2015
```

onoff version is 1.0.3.

fivdi, 2015-12-20:
Was onoff installed with one version of Node.js and used with a different version Node.js? If so, this isn't a good idea. In this case please reinstall onoff with `npm install onoff`.

If the issue persists, please post the complete output of the following commands:
```
node -v
npm -v
cd
mkdir testonoff
cd testonoff
npm install onoff
```

rampIc15, 2015-12-20:
Thanks!! for a quick response. 

I tried re-installing onoff but the issue is not resolved.

```
> epoll@0.1.16 install /home/pi/mytools/gateway/node_modules/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/mytools/gateway/node_modules/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory '/home/pi/mytools/gateway/node_modules/onoff/node_modules/epoll/build'
onoff@1.0.3 node_modules/onoff
└── epoll@0.1.16 (bindings@1.2.1, nan@2.1.0)
```
Here is the output requested : 
```

pi:/home/pi/mytools/test>node -v
v0.10.0

pi:/home/pi/mytools/test>npm -v
1.2.14

pi:/home/pi/mytools/test>npm install onoff
npm http GET https://registry.npmjs.org/onoff
npm http 304 https://registry.npmjs.org/onoff
npm http GET https://registry.npmjs.org/epoll
npm http 304 https://registry.npmjs.org/epoll
npm http GET https://registry.npmjs.org/bindings
npm http GET https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/bindings
npm http 304 https://registry.npmjs.org/nan

> epoll@0.1.16 install /home/pi/mytools/test/node_modules/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/mytools/test/node_modules/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory '/home/pi/mytools/test/node_modules/onoff/node_modules/epoll/build'
onoff@1.0.3 node_modules/onoff
└── epoll@0.1.16 (bindings@1.2.1, nan@2.1.0)

fivdi, 2015-12-20:
Node.js v0.10.0 and npm v1.2.14 are very very old! Where does this version of Node.js come from? I'm actually surprised that onoff appears to have installed successfully. Internally onoff depends on epoll which in turn depends on nan v2.1.x. nan v2.1.x supports Node.js 0.10.29+.

What's the output of the following commands?
```
cd /home/pi/mytools/test/node_modules/onoff/node_modules/epoll/test
./run-tests
```

Is there anything preventing you from using Node.js v4 or higher?

rampIc15, 2015-12-20:
yep. node is old. This Pi has been running fine for a while so did not bother to update it.

Anyhow I installed new version of node and npm. Re-installed onoff. Everything good now!.
Thanks for your help. And great job with onoff.

cheers !

closed 2015-12-20 by rampIc15

fivdi, 2015-12-20:
Thanks!
Have fun with onoff.