fivdi/onoff #44: Receive more than once the same button state

shayhz, 2016-03-08:
Hi,

I'm using you package to listen to button event and I noticed that sometimes (don't know how to 100% simulate it) I receive more than once the same value.
Below is my code.
When running it and start pressing the button, I get sometimes twice the value 1 (in a row).

button.watch(function(err, value) {
  console.log(value);
});

Are you aware of such issue ?

Btw, great package :)

Thanks !

fivdi, 2016-03-08:
I'd imagine you're seeing the effects of button bounce here. Is this possible? There are articles on the Web discussing the topic, for example, [here](http://www.cc.gatech.edu/~hadi/teaching/cs3220/02-2015fa/doc/debounce.pdf). There are both hardware and software solutions to the problem. One good software solution to the problem on Linux systems like the Raspberry Pi is to use the gpio-keys driver for interfacing with buttons. In general, I use the Node.js [gpio-button](https://github.com/fivdi/gpio-button) module for interfacing with momentary push buttons. Internally, the gpio-button module uses the gpio-keys device driver. It's usage is a little comples but it provides very good results.

fivdi, 2016-03-11:
@shayhz did the above information help?

fivdi, 2016-03-23:
Assuming that the issue was resolved with the help of the previous comment.

closed 2016-03-23 by fivdi