fivdi/onoff #45: Error: EBUSY: resource busy or locked, write

musukvl, 2016-04-26:
If my app accidently closed without led4.unexport(); so now I have **Error: EBUSY: resource busy or locked, write** error for each start. How to avoid it? Raspberry restart does not help.

```
pi@raspberry-black:~/dev/homesen $ node test-gpio.js
fs.js:706
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EBUSY: resource busy or locked, write
    at Error (native)
    at Object.fs.writeSync (fs.js:706:20)
    at Object.fs.writeFileSync (fs.js:1234:24)
    at new Gpio (/home/pi/dev/homesen/node_modules/onoff/onoff.js:82:8)
    at Object.<anonymous> (/home/pi/dev/homesen/test-gpio.js:3:12)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
```

```
var Gpio = require('onoff').Gpio;
var led4 = new Gpio(4, 'out');
blink(led4, 500);

function blink(led, period) {
    led.write(1, function () {
        setTimeout(function(){
            led.write(0);
        }, period);
    });
}
```

musukvl, 2016-04-26:
```
pi@raspberry-black:~/dev/homesen $ gpio readall
 +-----+-----+---------+------+---+---Pi 2---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5V      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 0 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 1 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 2---+---+------+---------+-----+-----+

```

fivdi, 2016-04-27:
Thanks for filing this issue. This shouldn't happen. I'll look into it asap.

musukvl, 2016-04-27:
I have raspberry pi 2 with this issue. So I can provide any debug info you need. Restart does not help, so looks like there is some lockfile, but I have no idea what can be its name.

fivdi, 2016-04-27:
The exact same test program works fine for me on a Pi 2:
```js
pi@raspberrypi:~/onoff $ cat test.js 
var Gpio = require('onoff').Gpio;
var led4 = new Gpio(4, 'out');
blink(led4, 500);

function blink(led, period) {
    led.write(1, function () {
        setTimeout(function(){
            led.write(0);
        }, period);
    });
}

pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ ls /sys/class/gpio/
export  gpio4  gpiochip0  unexport
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ ls /sys/class/gpio/
export  gpio4  gpiochip0  unexport
pi@raspberrypi:~/onoff $ node test.js 
pi@raspberrypi:~/onoff $ 
```
As can be seen, the test program can be called multiple times without issue.

This indicates that GPIO4 on your system is being reserved by some other software and is therefore not available for usage in the test program that turns the LED on and off. This can happen when device tree overlays are loaded. For example, if the device tree overlay `w1-gpio-overlay.dtb` is loaded by adding the following line to `/boot/config.txt`
```
device_tree_overlay=overlays/w1-gpio-overlay.dtb
```
then GPIO4 will be reserved for 1-Wire devices. Running the test program will then result in the following output (which is exactly what you're seeing):
```
pi@raspberrypi:~/onoff $ cat test.js 
var Gpio = require('onoff').Gpio;
var led4 = new Gpio(4, 'out');
blink(led4, 500);

function blink(led, period) {
    led.write(1, function () {
        setTimeout(function(){
            led.write(0);
        }, period);
    });
}

pi@raspberrypi:~/onoff $ node test.js 
fs.js:706
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EBUSY: resource busy or locked, write
    at Error (native)
    at Object.fs.writeSync (fs.js:706:20)
    at Object.fs.writeFileSync (fs.js:1234:24)
    at new Gpio (/home/pi/onoff/node_modules/onoff/onoff.js:82:8)
    at Object.<anonymous> (/home/pi/onoff/test.js:2:12)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
pi@raspberrypi:~/onoff $ 
```
Is there something running on your Pi 2 that is reserving GPIO4? For example, is a device tree overlay being loaded and resedving GPIO4?

musukvl, 2016-04-27:
I don't think I have any other gpio apps running.
Bug I had `dtoverlay=w1-gpio` in `/boot/config.txt`.
I removed it and test-gpio.js app it become to work. 

Thanks for help.

fivdi, 2016-04-27:
No problem. Thanks for the feedback and have fun with onoff :)

closed 2016-04-27 by fivdi