fivdi/onoff #47: Reverse

thisboyiscrazy, 2016-05-01:
A lot of times the io is connected to relays the reverse the output. E.G. Gpio goes low, relay closes. It would be nice to have a "reverse" parameter in the constuctor.

fivdi, 2016-05-01:
So writing 0 to a GPIO sets it to 1 and writing a 1 sets it to 0? 0 would read as 1 and 1 would read as 0?

thisboyiscrazy, 2016-05-01:
Exactly.

fivdi, 2016-05-03:
This should also be possible for inputs. 0 would read as 1 and 1 would read as 0. rising edge interrupts would become falling edge interrupts and falling edge interrupts would become rising edge interrupts.

fivdi, 2016-05-03:
This is a good idea. I'll add it.

thisboyiscrazy, 2016-05-04:
Excellent Thanks

fivdi, 2016-05-04:
onoff v1.1.0 was just published on npm. The Gpio constructor has an new option called `activeLow` that can be used to get the desired functionality:
```js
var Gpio = require('onoff').Gpio,
  input = new Gpio(7, 'in', {activeLow: true}),
  output = new Gpio(8, 'out', {activeLow: true});
```

closed 2016-05-04 by fivdi

fivdi, 2016-05-06:
You're welcome :)
Have fun with onoff.