fivdi/onoff #48: how to read value from IR obstacle sensor

m-devan, 2016-05-20:
hi
im not getting correct value when using read(err,value) or readSync functions.
im using Gpio 3 for IR sensors output
pls help
thanks in advance

fivdi, 2016-05-20:
On which type of system are you having the issue, a Raspberry Pi?

If it is a Raspberry Pi, is it GPIO3 on pin 5 of the [GPIO header](http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png)?

Can you post a link to a description of the IR obstacle sensor being used?

Can you post the code which has the issue?

m-devan, 2016-05-21:
var Gpio = require('onoff').Gpio,
  pir = new Gpio(3, 'in');

pir.read(function(value) {
      console.log(value);
});
++++++++++++++++++++
always returns null value
+++++++++++++++++++++++
IR SENSOR: http://www.ebay.in/itm/331669024235?aff_source=Sok-Goog

with python it works fine

fivdi, 2016-05-21:
I'm assuming that we're talking about a Raspberry Pi here and that GPIO3 is Pin 5 on the [GPIO header](http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png).

The [read](https://github.com/fivdi/onoff#readcallback) completion callback is passed two arguments, err and value. Try with the following code:
```js
var Gpio = require('onoff').Gpio,
  pir = new Gpio(3, 'in');

pir.read(function(err, value) {
  if (err) {
    throw err;
  }

  console.log(value);
});
```

m-devan, 2016-05-21:
yes it raspi, with the above code value is always 1. 
even when we have an object 
++++++++++++++++++++
working python code is
import RPi.GPIO as GPIO
import time
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BOARD)
GPIO.setup(3, GPIO.IN)
count = 0                            
while True:
          i=GPIO.input(3)                  
          if i==1:                               
               print "Obstacle detected ",count
               count += 1
               time.sleep(0.1)

thanks for your help

fivdi, 2016-05-21:
There are multiple ways of numbering the pins on the Raspberry Pi GPIO header. The above Python program uses the BOARD numbering system. There is also the BCM numbering system. onoff uses the BCM numbering system. The above Python program **IS NOT USING GPIO3,  IT'S USING GPIO2**

**Please** look at this image of the GPIO header: http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png

Is it correct that the IR obstacle sensor is connected to GPIO2, which is pin 3 on the GPIO header?

If this is correct, please try with the following program:
```js
var Gpio = require('onoff').Gpio,
  pir = new Gpio(3, 'in');

pir.read(function(err, value) {
  if (err) {
    throw err;
  }

  console.log(value);
});
```

m-devan, 2016-05-21:
yes, thanks a lot changing to use GPIO 2 in code works fine. sorry for the trouble. i am new to this and programming as well.
the above code reads current value correctly. i need to run the program continuously and  if value is 1 (intruder alert) i need to print and continue sensing.....how to modify the program
thanks for your help

fivdi, 2016-05-21:
Good that you have it working now :)

To perform a task at regular intervals, `setInterval` can be used. For example, to read the sensor every 100 milliseconds, try the following:

```js
var Gpio = require('onoff').Gpio,
  pir = new Gpio(2, 'in');

setInterval(function () {
  pir.read(function(err, value) {
    if (err) {
      throw err;
    }

    console.log(value);
  });
}, 100);
```

I'm going to close this issue now as the initial issue has been resolved.

closed 2016-05-21 by fivdi

m-devan, 2016-05-21:
thanks a lot

fivdi, 2016-05-21:
You're welcome :)
Have fun with onoff.