fivdi/onoff #52: EPERM: operation not permitted, write

SimonTheEngineer, 2016-07-23:
Hi, I'm just trying the functions out and I'm currently trying to write to pin 4, but getting the above error message. Here's the code, it's pretty basic..

`var Gpio = require('onoff').Gpio,
  led = new Gpio(4, 'out');

  led.write(1);`

Has anyone come across this before?

SimonTheEngineer, 2016-07-23:
My apologies for the double submit..

This was solved by running the script with SUDO

closed 2016-07-23 by SimonTheEngineer

fivdi, 2016-07-23:
@SimonTheEngineer if Raspbian Jessie is being used as mentioned [here](https://github.com/fivdi/onoff/issues/53) then it should not be necessary to run the script with sudo if logged on as user `pi`. Were you logged on as user `pi`?

SimonTheEngineer, 2016-07-23:
I was yes, fresh install too so all permissions should be default

fivdi, 2016-07-23:
I just gave it a try with Raspbian Jessie and don't have the issue. i.e., it's not necessary to run the script with sudo.

What happens if you run the following commands from the command line? They should turn the LED connected to gpio4 on. Do the commands turn the LED on or is there an EPERM error?

```
pi@raspberrypi:~ $ cd /sys/class/gpio/
pi@raspberrypi:/sys/class/gpio $ echo 4 > export 
pi@raspberrypi:/sys/class/gpio $ echo out > gpio4/direction 
pi@raspberrypi:/sys/class/gpio $ echo 1 > gpio4/value 
pi@raspberrypi:/sys/class/gpio $ 
```

fivdi, 2016-07-23:
Note that `EPERM: operation not permitted, write` isn't actually related to permissions. An `EPERM` error occurs when an operation is not permitted because it doesn't make sense. In this case it looks like the `EPERM` error occurred because an an attempt was made to write data to an input. Here's an example:
```
pi@raspberrypi:~ $ cd /sys/class/gpio/
pi@raspberrypi:/sys/class/gpio $ echo 17 > export 
pi@raspberrypi:/sys/class/gpio $ cd gpio17
pi@raspberrypi:/sys/class/gpio/gpio17 $ echo in > direction 
pi@raspberrypi:/sys/class/gpio/gpio17 $ echo 1 > value 
-bash: echo: write error: Operation not permitted
pi@raspberrypi:/sys/class/gpio/gpio17 $ 
```
Errors related to permissions are usually `EACCES` errors.