fivdi/onoff #54: Does not work anymore on Node v4.5.0 (ARM)

neophob, 2016-09-02:
The latest onoff version does not work anymore on my arm machine. this is the simple code is use:

```
const GpioOnOff = require('onoff').Gpio;
const button = new GpioOnOff(239, 'in', 'both');
button.watch(function(err, value) {
  console.log('jo');
  console.log(err, value);
});
```

-> the button watch function is never triggered.

when reading directly `cat /sys/class/gpio/gpio239/value` is see that the values are populated correctly.

any hints?

fivdi, 2016-09-02:
What type of ARM machine is being used?
Did the same code work in the past?

neophob, 2016-09-02:
yes, its a armv7, i just updated node from v4.4.x to v4.5.0

fivdi, 2016-09-02:
I can't reproduce the error with the sample program posted above using Node.js 4.5.0 and GPIO4 on an ARMv7 Raspberry Pi 2:
```
pi@raspberrypi:~/onoff $ node -v
v4.5.0

pi@raspberrypi:~/onoff $ cat t.js 
const GpioOnOff = require('onoff').Gpio;
const button = new GpioOnOff(4, 'in', 'both');
button.watch(function(err, value) {
  console.log('jo');
  console.log(err, value);
});

pi@raspberrypi:~/onoff $ node t.js 
jo
null 1
jo
null 0
jo
null 1
jo
null 0
jo
null 1
jo
null 0
```

> i just updated node from v4.4.x to v4.5.0

Is Node.js the only thing that was updated or were other things updated, for example, the operating system? If the operating system was updated, is it possible that the GPIO driver was updated and that the old driver supported interrupt detection on GPIO239 but the new driver doesn't?

fivdi, 2016-09-02:
What type of computer is being used? As in Raspberry Pi, BeagleBone, CHIP, ...

neophob, 2016-09-05:
I use an ARM A20 CPU, like the CHIP uses. its very strange, because it seems that the `watch` function is not called always. sometimes it works, but not really deterministic. when I run `button.readSync();` it works aswell:

```
> button
Gpio {
  gpio: 239,
  gpioPath: '/sys/class/gpio/gpio239/',
  opts: { debounceTimeout: 0 },
  readBuffer: <Buffer 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>,
  listeners: [ [Function] ],
  valueFd: 15,
  poller: Epoll { closed: false } }
> button.readSync();
0
> button.readSync();
1
```

.. any hints?

neophob, 2016-09-05:
OK - I know what triggers the problem. I did already export the gpio in my application - now when I start my test above, the watch function almost never works. however if I run `echo 239 > /sys/class/gpio/unexport` I let onoff export the GPIO port, and THEN the watch function works.

Maybe it make sense to add an option that onoff should not touch the gpio config?

fivdi, 2016-09-06:
I can't reproduce the error by running an app that exports the GPIO and a second app that uses the GPIO.

Here's the app that exports the GPIO:
```js
const GpioOnOff = require('onoff').Gpio;
const button = new GpioOnOff(4, 'in', 'both');

setInterval(function () {
  console.log('Hello, World!');
}, 1000);
```
Here's the test app that uses the GPIO:
```js
const GpioOnOff = require('onoff').Gpio;
const button = new GpioOnOff(4, 'in', 'both');
button.watch(function(err, value) {
  console.log('jo');
  console.log(err, value);
});
```

Some questions posted above haven't been answered:

* Did the same code work in the past?
* Is Node.js the only thing that was updated or were other things updated, for example, the operating system?

The title of this issue is "Does not work anymore on Node v4.5.0 (ARM)". This title implies that it did work in the past. I'm almost certain that the update from Node.js v4.4.x to Node V4.5.0 didn't cause the issue and that something else was also modified but not mentioned yet.

In order to move forward sample code to reproduce the error is needed.

neophob, 2016-09-07:
Yes you're right, this bug report is completely borked. I create a new one.

closed 2016-09-07 by neophob