fivdi/onoff #56: v4.0.0 still gives compiling error: module did not self-register

haldunatar, 2016-09-17:
I updated gcc and g++ versions to 4.8 but It still gives error pointing bindings.js

I followed https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons

Any advice?
Thanks in advance!

fivdi, 2016-09-17:
What is the exact command being used to install `onoff` and what is the complete output of that command?

Also, what is the output of the following commands?

```
node -v
uname -a
gcc --version
```

haldunatar, 2016-09-17:
npm install onoff

V4.0.0

Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

gcc (Raspbian 4.8.4-1) 4.8.4

onoff was just working fine with 0.10.30. I updated node to 4.0.0 and If I run the file where onoff included is:

home/pi/Desktop/ledWithNode/anotherTest/raspberry-nodejs-simple-led/node_modules/onoff/node_modules/epoll/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: Module did not self-register.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at bindings (/home/pi/Desktop/ledWithNode/anotherTest/raspberry-nodejs-simple-led/node_modules/onoff/node_modules/epoll/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/pi/Desktop/ledWithNode/anotherTest/raspberry-nodejs-simple-led/node_modules/onoff/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)

fivdi, 2016-09-17:
Did you uninstall and reinstall `onoff` after updating Node.js from v0.10.30 to v4.0.0? It's essential to do this as `onoff` depends on a compiled native module called `epoll`. A native module compiled for Node.js v0.10.30 will not function with Node.js v4.0.0. It needs to be compiled for the version of Node.js being used.

haldunatar, 2016-09-17:
That makes sense and it works! Thank you!

closed 2016-09-17 by haldunatar

fivdi, 2016-09-17:
Your welcome. Have fun with `onoff`.

fivdi, 2016-09-17:
Additional information that may help others reading this issue: The "Module did not self-register" error was a runtime error, not a compile time error.