fivdi/onoff #57: All relays turn on when starting the app

docmattman, 2016-10-10:
I'm using onoff to control a relay board with LEDs.  When I start the app script, all LEDs turn on automatically.  Is there a way to prevent onoff from turning all my relays on when initializing them?

fivdi, 2016-10-10:
The direction argument passed to the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) is usually `'out'` for an output Gpio. Instead of `'out'`, `'high'` or `'low'` can also be specified and will configure the GPIO as an output with an initial level of high or low respectively.

Can `'high'` or `'low'` be specified to achieve the desired behavior?

fivdi, 2016-10-12:
@docmattman were you able to make any progress here?

danb5, 2016-10-13:
This bug makes this unusable.

fivdi, 2016-10-13:
So calling the Gpio constructor with a direction argument of `'high'` or `'low'` instead of `'out'` didn't help?

danb5, 2016-10-13:
I must admit to being a novice.  I am trying to use this with a home bridge plugin called homebridge-gpio-garagedoor.  It uses onoff as a dependancy for goop control.   Not sure how I change how onoff sets the pin.

danb5, 2016-10-13:
Forgot.  I am using a Raspberry Pi zero running Jesse.

danb5, 2016-10-15:
Where do I call the grip constructor

fivdi, 2016-10-15:
It looks link it's being called indirectly from here: https://github.com/KraigM/homebridge-gpio-garagedoor/blob/master/lib/SwitchPort.ts#L20

That's TypeScript code so the TypeScript transpiler will need to be called to generate the JavaScript code.

danb5, 2016-10-15:
Hmm,  I will need to figure out what you mean but I see where you are pointing in the code.  I can't thank you enough for the help.

fivdi, 2016-11-14:
@docmattman @danb5 were you able to make any progress here?

danb5, 2016-11-14:
Sorry, yes you were correct in the issue. 

Dan

fivdi, 2016-11-15:
That's good news. Thanks for the feedback.

closed 2016-11-15 by fivdi

docmattman, 2016-12-06:
Sorry for the late response.  Setting the direction to `high` or `low` did work.  At the time, I wasn't aware those were options.  I only saw `out`.  So, yes, problem solved.  Thanks!

fivdi, 2016-12-06:
I'm glad it worked and thanks for taking the time to provide the feedback.

Cirrostratuss, 2018-01-02:
@docmattman where did you change the direction?