fivdi/onoff #58: Clocking signal

rogueturnip, 2016-11-07:
I wanted to find out if onoff would be a good solution for watching a clock signal on a pin.  What I have is a clock that runs for a period of time and a data line that can be read on the clock high and written to on clock low.  After a period of time it goes all high for about 500ms to synchronize and then starts again.

Would onoff be a good option to monitor this clock so I could read or write the data?

Thanks!

fivdi, 2016-11-07:
What's the frequency of the clock signal during the data transfer phase and what type of computer is being used?

rogueturnip, 2016-11-07:
I'm hoping to drive this off a beaglebone.  I've attached what a scope shows of the clocking.
[image]
[image]

fivdi, 2016-11-07:
That looks like a 1KHz clock. There's a reasonable chance that it will work.

fivdi, 2016-12-06:
@rogueturnip was it possible to watch the clock signal with onoff?

fivdi, 2016-12-24:
On second thoughts, this is very unlikely to function correctly. A 1KHz clock signal means 2000 state changes per second which in turn implies an interrupt every 500 microseconds. Internally onoff has a thread that calls epoll to detect interrupts. When an interrupt is detected by this thread, the appropriate callback in the JavaScript main thread is called. Given that Linux isn't a realtime operating system interrupts are almost certain to be missed at this rate.

closed 2016-12-24 by fivdi