fivdi/onoff #59: Detecting long press for buttons

emrekilic, 2016-11-13:
I am using this code block for detecting 5 sec long press into button. This code block can detect long press. But after some time like 2 days later, it detect other long pressed when i am sure that nobody pressed the button. Are there anything wrong in code block? Any advice for detecting long press.

```
var BUTTON_CONTROLLER = false;
var buttonControllerTimer;
var button = new Gpio(27, 'in', 'both');

button.watch(function (err, value) {
    if (BUTTON_CONTROLLER && value == 1) {
                console.log('button long pressed');
    }
    BUTTON_CONTROLLER = false;
    clearTimeout(buttonControllerTimer);
    buttonControllerTimer = setTimeout(buttonClear, 5000);
});

function buttonClear() {
    BUTTON_CONTROLLER = true;
}
```

fivdi, 2016-11-13:
If the button wasn't pressed at all after 2 days, not even shortly, the signal indicating that it was pressed must have been generated some other way. For example, touching the corresponding circuit or touching wires on breadboards can result in such signals. Could something like this have happened?

emrekilic, 2016-11-13:
First i tought as you said and i put pcb inside a box. So nobody can touch pcb. But problem occured again. After that i thought it could be an software problem.

fivdi, 2016-11-13:
It's difficult to believe that it could be a software issue. I don't see how the watch callback could be called with the state of the button changing.

Which GPIO is being used for the button and is a pulldown resistor being used?

emrekilic, 2016-11-13:
I posted code block in my first comment. You can see which gpio used and how i used watch callback.
I used GPIO27 but i didnt use any resistor. i thought that GPIO27 is already pulldown. Because when i log watch callback it was really stable and it didnt get any wrong results. I never think, it was about your library but i didnt find the mistake which i did.

fivdi, 2016-11-14:
Sorry about the delay in answering. Yes, GPIO27 is pulldown by default. It might be the library, I don't know. It's just that I'm not aware of any code in the library that would result in the watch callback being called without the state of GPIO27 changing.

Can you try with an external pulldown resistor of about 10K? The internal pulldown resistors are about 50K which is fairly week. A stronger external pulldown might fix things.

fivdi, 2016-12-06:
@emrekilic were you able to try with a stronger external pulldown and if so, did it help?

emrekilic, 2016-12-06:
I am really busy in this days and did not try it. But when i check my database, 3 times long press detected in this week. I am sure that nobody press it.

fivdi, 2016-12-23:
This is most likely a hardware problem so I'm going to close the issue. If it turns out to be a problem with onoff the issue can be reopened.

closed 2016-12-23 by fivdi