fivdi/onoff #60: Infinit watch() calls

mathiask88, 2016-12-07:
Hey,

I got a normally closed switch connected to GPIO22 on a RPI3. The Pin voltages are 2960mV in closed state and 20mV in opened state, so everything seems fine. But if I watch the switch with
```javascript
const referenceSwitchMeasure = new Gpio(22, 'in', 'falling');

referenceSwitchMeasure.watch((err, value) => {
  console.log('Switch value: ' + value);
});
```

I get a single line for `Switch value: 0` but infinit `Switch value: 1` :(

`Linux raspberrypi 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux`
`Node.js 4.5.0`
`onoff v1.1.1`

mathiask88, 2016-12-07:
I also don't know why that example does interrupt for falling and rising edges?!

fivdi, 2016-12-17:
Does this happen without opening the switch or must the switch be opened to see this behavior?

Is the switch really connected to pin 15 (GPIO22) on the GPIO header and not to pin 22 (GPIO25)?
Is an external pull-down resistor resistor being used? The internal pull-down resistor which is configured by default for GPIO22 may be too weak.

The example watches for falling and rising edges in order to turn the LED on when the button is pressed and turn it off when the button is released.

fivdi, 2016-12-28:
@mathiask88 were you able to make any progress here?

pandrew1, 2017-01-03:
I'm seeing similar behavior with watch events firing unexpectedly when there's no change in the input. It happens apparently randomly a few times an hour and I've updated my code to ignore it if there's been no change since the last event. Mostly I see them on 0 since the doors are mostly closed.

It seems to happen on a smaller kind of reed switch I have and not the larger roller door ones. I wondered if it was just wind or something and these are momentarily going open but too quick for the value on the event to show it. I only ever get one event, duplicating the previous event, and not a pair of events like it had two edges being detected.

My hardware is reed switches attached to doors which ground the GPIO inputs and there's 10K pull-up resistors on each input. I'm using GPIO 5, 6, 13, and 26. All are configured in onoff as this code snippet: new Gpio(5, 'in', 'both')

I'm running OnOff 1.1.1, Node.js v7.2.1, Raspberry Pi Linux 4.4.34-v7+ #930 SMP

fivdi, 2017-01-04:
@pandrew1 are you 100% sure that events are being fired with no change in the input? Do you have the possibility to attach an oscilloscope to verify whether or not the input is changing?

mathiask88, 2017-01-04:
@fivdi No not yet. The hardware is in the university, maybe next week. I definitely will try an oscilloscope.

pandrew1, 2017-01-06:
@fivdi I don't have a scope. These are the reed switches I'm using. There's a small air gap between them around the door so I can improve that but the door is not being opened when I get the extraneous firing of the event. I can excluded these easy enough because the value is the same as the previous event. https://www.amazon.com/gp/product/B015NI9W20

fivdi, 2017-01-06:
I think the issues here are hardware issues rather than software issues. If they are hardware issues finding their source isn't always easy. #61 is similar to this issue.

@pandrew1 I would imagine that you are seeing short spikes on the digital inputs. They're probably there for only a few microseconds. This would explain why the the value of the input doesn't appear to change. Internally onoff uses a module called epoll to detect hardware interrupts. When a interrupt is detected onoff is informed that the interrupt occurred but it's not informed about the value of the input when the interrupt occurred. It needs to read the digital input to determine what the value is. It will take a number of microseconds before onoff gets around to reading the digital input. By the time onoff gets around to reading the digital input it's value may have changed again, for example, due to a short spike on the input.

@mathiask88 please keep us updated on how things go.

fivdi, 2017-01-26:
This is most likely a hardware problem so I'm going to close the issue. If it turns out to be a problem with onoff the issue can be reopened.

closed 2017-01-26 by fivdi