fivdi/onoff #61: PIR sensor triggered with motion when nginx serves an HTTP request

daviddahl, 2016-12-24:
I am grasping at straws here - and don't doubt this error is not onoff's fault, but every time I load an HTTP request via nginx (latest raspian/debian on raspberry pi 3 B) requesting a JPG file, the motion sensor is triggered. 

If anyone has any idea how to begin to debug this I would appreciate it. So far I have isolated the code down to my motion detection module that merely does:

```
let Gpio = require('onoff').Gpio,
    pir = new Gpio(4, 'in', 'both');

function motionWatcher (err, value) {
    if (!value) {
        return console.info('PIR reset');
    }
    pir.unwatch(); // remove event listener to prevent too many reports
    // ... Do some reporting, etc....
    console.info('motion recorded');
    pir.watch(motionWatcher);
}

pir.watch(motionWatcher);

process.on('SIGINT', function () {
    pir.unexport();
});

```

daviddahl, 2016-12-24:
... As soon as the nginx request is done, the PIR lights up with a 1. This happens with no one in the room, or even in the house.

fivdi, 2016-12-24:
Maybe the power supply for the Pi isn't strong enough.

daviddahl, 2016-12-24:
I have one from Canakit which is 5v 2.5 amp. I have ordered a 5v 3 amp version. Thanks for the tip. Could it be that the voltage chnages significantly when nginx loads a JPG? I can login to nginx and load a `fancy_index` without any motion triggers.

fivdi, 2016-12-24:
I'd expect 5V 2.5A to be enough and wouldn't expect a significant voltage change when nginx loads a JPG.

fivdi, 2016-12-24:
Can you post a link to type of PIR being used?

fivdi, 2016-12-24:
Is there additional hardware connected to the Pi?

fivdi, 2016-12-24:
This issue reminds me of https://github.com/fivdi/onoff/issues/59.

daviddahl, 2016-12-24:
@fivdi AC adapter cam with this kit: https://www.amazon.com/gp/product/B01C6FFNY4/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1
the motion sensor came with this kit: https://www.amazon.com/gp/product/B0104JGLUE/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

I also have a Raspberry Pi 8 mbpxl camera attached that takes single frame pictures attached

fivdi, 2016-12-24:
That doesn't sound like too much hardware for the power supply.

On the "Technical Details" section of https://www.adafruit.com/product/189 it says: 
> PIR sensors and the Raspberry Pi 3 sometimes don't get along - if you're having false trigger reports, make sure the PIR Sensor is far away from the Pi 3.

Maybe the PIR sensor is too close to the Pi 3?

fivdi, 2017-01-06:
@daviddahl were you able to make any progress here?

fivdi, 2017-01-26:
This is most likely a hardware problem so I'm going to close the issue. If it turns out to be a problem with onoff the issue can be reopened.

closed 2017-01-26 by fivdi

daviddahl, 2017-01-26:
Yeah, sorry, I updated to a 3v Power Supply and the false alarms diminished greatly

fivdi, 2017-01-26:
Ok, thanks for the feedback. And I assume you mean a 3A/2.5A power supply :)

daviddahl, 2017-01-26:
oops. Yes! Amps not volts