fivdi/onoff #63: readSync does not return correct value on initial call.

gjws, 2017-01-26:
I'm using writeSync to query my gpio in order to confirm state prior to performing an operation.  Simple example:

```
var Gpio = require('onoff').Gpio
ssr = new Gpio(14, 'out')
console.log(ssr.readSync())
if (parseInt(ssr.readSync()) >= 1) {
        console.log("on " + ssr.readSync())
} else {
        console.log("off " + parseInt(ssr.readSync()))
}
ssr.writeSync(parseInt(process.argv[2]))
console.log(ssr.readSync())
```

The initial call on line 3 always returns 0. If I check the raw value in /sys/class/gpio/gpio14/value I can see it has a state of "1", however readSync returns a 0. It does however return the correct value in subsequent calls, such as line 10 above.

fivdi, 2017-01-26:
> The initial call on line 3 always returns 0.

Correct, this is the expected behavior. Constructing a new Gpio object with `new Gpio(14, 'out')` does a number of things. It exports GPIO14 to userspace, configures GPIO14 as a output AND sets the value of GPIO14 to 0. This is the documented behavior of the Linux [GPIO Sysfs Interface](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt).

> If I check the raw value in /sys/class/gpio/gpio14/value I can see it has a state of "1"

It's unclear to me when this is performed. Before running the above program or after running the above program? It doesn't really matter anyway. Constructing a new Gpio object with `new Gpio(14, 'out')` will always set the value of GPIO14 to 0.

>  It does however return the correct value in subsequent calls, such as line 10 above.

This is probably because it was set to the correct value on the previous line (`ssr.writeSync(parseInt(process.argv[2]))`).

The [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) has additional options that may be what you are looking for. For example, constructing a new Gpio object with `new Gpio(14, 'high')` exports GPIO14 to userspace, configures GPIO14 as a output and sets the value of GPIO14 to 1.

fivdi, 2017-01-26:
Thank you for reporting this issue. The documentation for the Gpio constructor was unclear and didn't state that new `Gpio(<number>, 'out')` would set the value of the corresponding GPIO to 0. The documentation has now been modified to make things clearer.

gjws, 2017-01-26:
Thank you for the clarification.

fivdi, 2017-01-26:
You're welcome.

closed 2017-01-26 by fivdi

fivdi, 2017-01-26:
One more thing. The readSync method returns a number so `parseInt(ssr.readSync())` can be simplified to `ssr.readSync()`.

gjws, 2017-01-26:
Thank you. I'm using parseInt to convert the command line parameter, then I got lazy with copy and paste elsewhere :)