fivdi/onoff #64: Example script not working..

patrick0888, 2017-01-30:
Hi,

When I run the example script of this repo on my raspberry pi 3 its doing nothing.
It looks like the gpio pins are not activated.

But after start and close a python script the gpiopins in the nodejs script start working.

Do I have to enable something before using the pins in de nodejs script?

fivdi, 2017-01-31:
Please post the code that can be used to reproduce the issue. Also, is an external pull-down resistor being used for the button?

fivdi, 2017-01-31:
It shouldn't be necessary to enable anything else for the example program to function correctly on  Raspberry Pi running Raspbian as user pi.

patrick0888, 2017-01-31:
Python
```
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)
GPIO.setup(26, GPIO.IN, pull_up_down=GPIO.PUD_UP)
```

Node

```
var Gpio = require('onoff').Gpio,
	button = new Gpio(26, 'in', 'both');
```

fivdi, 2017-01-31:
Here's what I think is happening with the program using onoff after the Pi boots.

One side of the button is connected to 0V and the other side of the button is connected to GPIO26. There is no external pull-up resistor pulling GPIO26 to 3.3V. The internal pull-down resistor for GPIO26 is activated and pulls GPIO26 to 0V. This is the default configuration for GPIO26 on the Pi. See page 102 of the [BCM2835 ARM Peripherals](https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) for further information. So, when the button is pressed, it connects GPIO26 to 0V, but the state of GPIO26 doesn't change as it's already 0V due to the activated internal pull-down resistor.

The Python program is different as it explicitly activates the internal pull-up resistor on GPIO26 and everything works as expected.

When the Python program terminates, the internal pull-up resistor on GPIO26 remains activated. If the onoff program is now run, it functions as expected.

There are a number of potential solutions here.

1) Connect the button to a GPIO that has it's internal pull-up resistor activated by default. See page 102 of [BCM2835 ARM Peripherals](https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) to determine which GPIOs have their internal pull-up resistors activated by default.
2) Use an external pull-up resistor for GPIO26 (the internal pull-up resistor is very weak and not really suitable for buttons in many scenarios.)
3) Use [pigpio](https://github.com/fivdi/pigpio/#buttons-and-interrupt-handling) rather than onoff as pigpio allows the internal pull-up/pull-down resistors to be configured at runtime from JavaScript,
4) Use onoff and device tree overlays to configure the internal pull-up/pull-down resistors as required. See [Configuring pullup and pulldown resistors](https://github.com/fivdi/onoff#configuring-pullup-and-pulldown-resistors) for further information.

fivdi, 2017-02-01:
Closing as there's no TODO for onoff here. If it turns out that it is a problem with onoff the issue can be reopened.

closed 2017-02-01 by fivdi

patrick0888, 2017-02-01:
Thanks I will try it out.