fivdi/onoff #65: EBADF, bad file descriptor

avonmoll, 2017-02-16:
I get the following error when I <kbd>Ctrl</kbd>+<kbd>c</kbd> to terminate my alarm clock script running on RPI 3 under PIXEL:

```
fs.js:469
  return binding.close(fd);
                 ^
Error: EBADF, bad file descriptor
    at Error (native)
    at Object.fs.closeSync (fs.js:469:18)
    at Gpio.unexport (/home/alex/Documents/pi-clock/node_modules/onoff/onoff.js:340:6)
    at process.<anonymous> (/home/alex/Documents/pi-clock/pi-clock.js:97:14)
    at process.emit (events.js:104:17)
    at Signal.wrap.onsignal (node.js:655:46)
```

I can't tell if my code is causing the issue or if it's an issue with `onoff`, itself. I followed this approach from the README (with `led` and `button` replaced with the `Gpio` objects I created):

```javascript
process.on('SIGINT', function () {
  led.unexport();
  button.unexport();
});
```

fivdi, 2017-02-16:
If an attempt is made to close a file and an "EBADF, bad file descriptor" error is thrown, it means that the file descriptor isn't a valid open file descriptor.

Is it possible that the `unexport` method is being called more than once on the same Gpio object?

Can you post a complete example that is as short as possible which can be used to reproduce the error?

fivdi, 2017-02-16:
Here's a test program which can be used to reproduce the issue. It calls `unexport` twice on the same Gpio object.
```js
"use strict";

var Gpio = require('onoff').Gpio,
  led = new Gpio(46, 'out');

led.unexport();
led.unexport();
```
Running this program on a BeagleBone Black will result in the same error:
```
fs.js:518
  return binding.close(fd);
                 ^

Error: EBADF: bad file descriptor, close
    at Error (native)
    at Object.fs.closeSync (fs.js:518:18)
    at Gpio.unexport (/root/onoff/node_modules/onoff/onoff.js:340:6)
    at Object.<anonymous> (/root/onoff/EBADF.js:7:5)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
```

fivdi, 2017-02-16:
A note has been added to the [documentation for the unexport method](https://github.com/fivdi/onoff#unexport) specifying that a Gpio object should not be used after calling its unexport method.

avonmoll, 2017-02-17:
You made some good points -- calling `unexport()` on previously `unexport()`ed `Gpio` objects is precisely what was happening. I just didn't realize that was the case until I stumbled across http://stackoverflow.com/a/9828572/6066890 -- I was hitting <kbd>Ctrl</kbd>+<kbd>c</kbd> twice to terminate my script without even thinking about it. Disclaimer: I'm new to raspberry pi, linux (in general), and Node.js.

Here's a simple workaround, but it seems kind of hacky:
```javascript
process.on("SIGINT", function() {
    // do stuff
    led.unexport();
   process.exit();
}
```

closed 2017-02-17 by avonmoll

fivdi, 2017-02-17:
I don't think hitting ctrl+c twice can result in the SIGINT handler being called twice.

avonmoll, 2017-02-18:
I tried to do some more searching on this and I found a [SO question and answer](http://stackoverflow.com/questions/14372288/how-to-properly-handle-sigint-with-express-js) which roughly correlates to the behavior I observed and the "fix" I had to implement.

fivdi, 2017-02-18:
Ok, I think I know what's going on now. The example program in the readme is fairly simple:

```js
var Gpio = require('onoff').Gpio,
  led = new Gpio(14, 'out'),
  button = new Gpio(4, 'in', 'both');

button.watch(function (err, value) {
  if (err) {
    throw err;
  }

  led.writeSync(value);
});

process.on('SIGINT', function () {
  led.unexport();
  button.unexport();
});
```
The only reason this program doesn't terminate directly after it's stated is because `button.watch` was called which means the program is asynchronously waiting in the background for the button to be pressed.

If no 'SIGINT' handler is installed, the default behavior of Node.js is to terminate the program when ctrl+c is hit. If a 'SIGINT' handler is installed the default behavior is removed and the program will not terminate when ctrl+c is hit. See [Signal Events](https://nodejs.org/docs/latest-v0.10.x/api/process.html#process_signal_events).

In the simple example above, the 'SIGINT' handler calls `button.unexport()`. This stops the program from asynchronously waiting in the background for the button to be pressed. This in turn that means that the program has nothing left to do, not even wait for the button to be pressed, so it terminates.

Your use case is most likely more complex so there will be additional asynchronous code running in the background which prevents the program from terminating whe ctrl+c is hit. Calling `process.exit()` tells Node.js to terminate the process as quickly as possible even if asynchronous code is running in the background.

avonmoll, 2017-02-19:
This makes total sense to me! My original code (and even my tiny test script that reproduces the behavior) was scheduling events in the future using 'setTimeout'. After the first SIGINT is captured it's likely still waiting on these, thus a second Ctrl + c is required. Either way, adding 'process.exit()' seems safe and easy for my application. 

Thanks for digging into this issue (and thanks for the sweet package!).