fivdi/onoff #66: wrong value for input status

rafaelquines, 2017-02-18:
Hi, congratulations by your package.

I'm using it to monitor a shield that I developed to a Raspberry 3. It consists basically in 14 inputs with Optocouplers. To test, I connected a simple switch in my shield.

Today, this shield already is used by some customers, but initially I develop a C#.NET with Mono to monitor theses status. I'm working to migrate everything to nodejs. Based on it, I believe it can't be a schematic error.

I'm using a debounceTimeout of 15ms.

While using your package, sometimes, the "watch" event is called, because I changed the switch, but the value of input received by my function is wrong. It is 0 when must be 1.

Take a look in my app logs:

2017-02-18T14:24:30.441Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:24:30.809Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:24:31.417Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:24:33.515Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:24:33.727Z - info: EVENTO - Input: Input5 Status: 0
**2017-02-18T14:24:33.995Z - info: EVENTO - Input: Input5 Status: 0**
2017-02-18T14:24:34.182Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:24:34.353Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:24:34.752Z - info: EVENTO - Input: Input5 Status: 0

Again

2017-02-18T14:26:37.731Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:26:38.776Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:26:40.343Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:26:41.273Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:26:42.076Z - info: EVENTO - Input: Input5 Status: 0
**2017-02-18T14:26:42.883Z - info: EVENTO - Input: Input5 Status: 0**
2017-02-18T14:26:45.790Z - info: EVENTO - Input: Input5 Status: 0
2017-02-18T14:26:47.756Z - info: EVENTO - Input: Input5 Status: 1
2017-02-18T14:26:48.752Z - info: EVENTO - Input: Input5 Status: 0

My module:

```
function InputMonitor(pin, name, debounce) {
    this._pin = pin;
    this._name = name;
    this._debounce = debounce ? 0 : debounce;
    this._input = new Gpio(pin, 'in', 'both', {
        debounceTimeout : debounce
    });
}
InputMonitor.prototype.start = function() {
    var that = this;
    this._input.watch(function (err, value) {
        if(typeof that._callback === 'function')
            that._callback(err, that._name, value);
    });
};
InputMonitor.prototype.onInputChange = function(callback) {
    this._callback = callback;
}
```

Usage:

```
var in5 = new InputMonitor(9, 'Input5', 15);
in5.onInputChange(function(err, name, value) {
    logger.info('EVENTO - Input: ' + name + ' Status: ' + value);
});
in5.start();
```
Can you have some idea about what's happening?

Thanks

fivdi, 2017-02-18:
Here is what I think is happening: Internally, onoff uses the linux epoll API to monitor state changes on GPIO pins. When the state of a GPIO pin changes, onoff is notified that the state of the pin changed. However, it is not notified to what state the pin changed. In order to determine the state of the pin, the value of the pin needs to be read. A certain amount of time passes between being notified about the state change and reading the value of the pin. During that time, the value of the pin changes again due to bounce. 

Potential solutions to this issue which may work for yur are discussed at #51.

rafaelquines, 2017-02-18:
Thanks @fivdi 
I did some tests with the GottZ solution.

https://gist.github.com/GottZ/e4be78c928817c6701155f6ae910d841

At first, it works well. I'll do more tests.

Thank you so much.

closed 2017-02-18 by rafaelquines

fivdi, 2017-02-18:
> Thanks @fivdi

@rafaelquines You're welcome!