fivdi/onoff #67: write 1 to GPIO before it gets defined

SamDecrock, 2017-03-21:
Hi,

I've been using your library for over 3 years now (to control our garage door). I used to make my own circuits but recently I bought one of those cheap Chinese relay boards.

Problem is that these relay outputs are activeLow. 

When I create a new GPIO using your library and set `activeLow` to `true` and then do `.writeSync(0);` right after after its creation, the relay turns on for a faction of a second.

Is it possible to send a 1 to the GPIO before it is defined as an output?

Sam

fivdi, 2017-03-21:
Hi Sam, yeah, you've being using onoff for quite a while :) It should be possible to achieve this by setting the direction argument that's passed to the Gpio constructor to 'high' rather than 'out'. See https://github.com/fivdi/onoff#gpiogpio-direction--edge--options

SamDecrock, 2017-03-21:
Thanks! I should take more time to read readme's...  :-)

closed 2017-03-21 by SamDecrock