fivdi/onoff #68: Wont work with other gpio library to control neopixel

guatedude2, 2017-04-10:
Hi, I'm trying to control a neopixle via [node-rpi-ws281x-native](https://github.com/beyondscreen/node-rpi-ws281x-native) without success. It seems to have some type of conflict.

Here's the issue *node-rpi-ws281x-native* side:
https://github.com/beyondscreen/node-rpi-ws281x-native/issues/55

fivdi, 2017-04-10:
Hi, from the [issue linked to above ](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/55) I can see that it was possible to get rpi-gpio to work. Given that rpi-gpio and onoff are actually quite similar I'd expect onoff to work also. Can you post an onoff program that doesn't work? If possible, the onoff equivalent of [the program you posted here](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/55#issuecomment-292852031).

guatedude2, 2017-04-11:
Sure, like in the other example I used a push button on GPIO24/ PIN 18 of my RPi and the strip of NeoPixles on 

```js

var ws281x = require('rpi-ws281x-native');

var NUM_LEDS = parseInt(process.argv[2], 10) || 10;
var pixelData = new Uint32Array(NUM_LEDS);

ws281x.init(NUM_LEDS);

var Gpio = require('onoff').Gpio;
var button = new Gpio(18, 'in', 'both');

button.watch(function(err, value) {
  if (err) throw err;
  console.log('button value is now ' + value);
  if (value) {
    newColor();
  }
});

// ---- trap the SIGINT and reset before exit
process.on('SIGINT', function () {
  ws281x.reset();
  button.unexport();
});

var offset = 0;
function newColor() {
  offset = Math.random()*1000;
  for (var i = 0; i < NUM_LEDS; i++) {
    pixelData[i] = colorwheel((offset + i) % 256);
  }

  offset = (offset + 1) % 256;
  ws281x.render(pixelData);
};

console.log('Press <ctrl>+C to exit.');

// rainbow-colors, taken from http://goo.gl/Cs3H0v
function colorwheel(pos) {
  pos = 255 - pos;
  if (pos < 85) { return rgb2Int(255 - pos * 3, 0, pos * 3); }
  else if (pos < 170) { pos -= 85; return rgb2Int(0, pos * 3, 255 - pos * 3); }
  else { pos -= 170; return rgb2Int(pos * 3, 255 - pos * 3, 0); }
}

function rgb2Int(r, g, b) {
  return ((r & 0xff) << 16) + ((g & 0xff) << 8) + (b & 0xff);
}
```

This is the output when I push down the button a number of times. 

```bash
# node test2.js
Press <ctrl>+C to exit.
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
button value is now 0
```

The value is never 1 and even if I

fivdi, 2017-04-11:
The `gpio` argument passed to the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) is a GPIO number, not a pin number.

Try with:
```js
var button = new Gpio(24, 'in', 'both');
```
rather than:
```js
var button = new Gpio(18, 'in', 'both');
```
What I don't understand is why there was actually output on the terminal when the button was pressed. I wouldn't have expected any output as the incorrect argument was passed to the Gpio constructor.

fivdi, 2017-04-13:
I just gave this a try and the button state changes were detected:
```
$ sudo node t.js 
Press <ctrl>+C to exit.
button value is now 1
button value is now 0
button value is now 1
button value is now 0
button value is now 1
button value is now 0
button value is now 1
button value is now 0
```
The button was connected to GPIO4 (pin 7) rather than GPIO24 (pin 18) but that shouldn't make a difference. See [GPIO header](http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png).

guatedude2, 2017-04-13:
Yeah you're right! Major oversight on my end... Thanks for taking the time and looking into it.

closed 2017-04-13 by guatedude2