fivdi/onoff #69: Exception "Module did not self-register" when trying to import onoff

chris-schmitz, 2017-04-20:
I'm running into a bug when trying to import onoff into an electron project. 

I used the electron-vue template to scaffold out a quick electron application.

I'm trying to pull in the onoff module so I can start communicating between my node and my raspberry pi's GPIO pins. When I import the onoff module I get the error "Module did not self-register".

[image]

[image]

I dug around in the issues and wiki section of your project and found the article [where you note the need for g++ and gcc](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons), so I checked my node version along with my g++/gcc versions:

[image]

The end goal is to build the app to run on my Raspberry Pi 3. 

Is there something I'm doing wrong or some other configuration/installation I need?

fivdi, 2017-04-20:
I have never used Electron so the information I'm providing here may not be the best. 

onoff uses a native module called epoll to detect state changes on GPIO pins. However, the compilation process for native modules is different for Electron and Node.js. See [here](https://github.com/electron/electron/issues/1774#issuecomment-105779085) for further information.

> I'm trying to pull in the onoff module so I can start communicating between my node and my raspberry pi's GPIO pins. When I import the onoff module I get the error "Module did not self-register".

In addition, onoff was implemented for Linux systems like the Raspberry Pi and not for Macs and PCs. Even it's actully possible to install onoff on a Mac or PC, it will not function correctly on a Mac or PC as they don't have GPIOs to talk to. If onoff is run on a Mac it would attempt to talk to GPIOs on the Mac, not the GPIOs on a Raspberry Pi.

chris-schmitz, 2017-04-22:
Thanks for the explanation and the link to the issue over in the electron repository. I'll read through that in a minute. 

> Even it's actully possible to install onoff on a Mac or PC, it will not function correctly on a Mac or PC as they don't have GPIOs to talk to. If onoff is run on a Mac it would attempt to talk to GPIOs on the Mac, not the GPIOs on a Raspberry Pi.

Yeah, I wasn't attempting to actually use onoff from my Mac. My Mac is my main development machine so that's where I'm building the UI and business logic for the app. My plan was to add onoff as a dependency to the project while I built out the UI on my Mac and then continue development on the GPIO portion from the Raspberry Pi. That way, most of the work could be done on my more powerful computer.

I'll try to figure out a different way of incorporating onoff. I want to keep it as a dependency in the same code base as the actual app, but I may end up creating a separate repository and then noting how to incorporate it in the main project. 

Thanks again!

fivdi, 2017-04-24:
Closing as there is no TODO for onoff.

closed 2017-04-24 by fivdi