fivdi/onoff #70: Infinit watch() calls after reboot

miwaniec, 2017-05-28:
Hi!

I have strange problem. After installation everything works ok, but after reboot it is not. After reboot, when I run my script, I don't get any information about pressed button (console.log in watch()). So I test button if it is ok. I ran python script with the same idea. It works.
I try again with node script and now I have infinity call watch with value 0 and 1.
What is wrong?
When I create new dir, install again library, it works.

I have code like this:

> var Gpio = require('onoff').Gpio,
>         led = new Gpio(18, 'out'),
>         button = new Gpio(23, 'in', 'both');
> led.writeSync(1);
> button.watch(function (err, value) {
>     console.log(value)
>     led.writeSync(value);
> });

Node v7.10.0
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux

fivdi, 2017-05-28:
If everything works after an install and no longer works after a reboot that is strange.

Maybe the issue is related to pull-up/pull-down resistors. As can be seen on page 102 of the [BCM2835 ARM Peripherals Datsheet](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) GPIO23 is pulled LOW with an internal pull-down resistor by default. Does your button circuit take this into account? Or does your button circuit use and external pull-up/pull-down resistor?

Can you post a photo of the circuit where the button and the wiring for the button is visible please?

miwaniec, 2017-05-28:
I used this button https://www.sparkfun.com/products/9181
When I close node script and run python script, I can read button state (but in python I check status in loop, not event handler).

fivdi, 2017-05-28:
I just tried the following program for a momentary push button connected to GPIO23. Everything functioned as expected. There were no infinite watch calls and everything still functioned after a reboot. The button has an external pull down resistor.
```js
var Gpio = require('onoff').Gpio,
  led = new Gpio(17, 'out'),
  button = new Gpio(23, 'in', 'both');

led.writeSync(1);

button.watch(function (err, value) {
  console.log(value)
  led.writeSync(value);
});
```
The button at https://www.sparkfun.com/products/9181 should work if connected up correctly in my opinion.

In the original post above I asked a number of questions that remain unanswered. Without this information I can't really provide any help.

 - As can be seen on page 102 of the [BCM2835 ARM Peripherals Datsheet](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) GPIO23 is pulled LOW with an internal pull-down resistor by default. Does your button circuit take this into account?
 - Or does your button circuit use and external pull-up/pull-down resistor?
 - Can you post a photo of the circuit where the button and the wiring for the button is visible please?

Note that you did provide a link to the button being used but this only describes the button and not the complete button circuit.

An additional question is

 - Is there any other software running on the Raspberry Pi that is using any of the pins on the GPIO header?

The fact that everything works initially after an install indicates that additional software may be running and this may be causing the issue.

fivdi, 2017-06-03:
@miwaniec any news on this one?

fivdi, 2017-06-07:
I'm going to assume that this is a hardware issue. If it turns out to be an issue with onoff this issue can be reopened.

closed 2017-06-07 by fivdi