fivdi/onoff #71: Change in GPIO output value causes multiple WATCH interrupts to fire

Vercellotti, 2017-06-05:
After many hours of pain and suffering, I finally resort to the community for help. Your assistance is greatly appreciated.

I am using a Raspberry Pi 3 with Raspbian (updated) and the latest version of Node .

I have 5 inputs (momentary switches) connected to the Rpi via an RC / schmitt trigger circuit and wired to 5 GPIO's. I have 5 outputs connected to an 8 channel relay which is powered separately from the RPi (different power sources).

When I press one of the momentary switches, it randomly causes multiple output GPIO's to trigger their relays - ie multiple unrelated GPIO's trigger when a single momentary switch is pressed. Somehow the individual WATCH methods appear to be activated when an unlinked GPIO is signalled.

When I physically disconnect the momentary switches from the Pi, the random WATCH process continues to trigger, but this happens way less often.

PLEASE assist me in decoding my problem, or guide me where I can look to identify the problem.

```
`var Gpio = require ('onoff').Gpio;

var deviceList = {'securityLight':{'code':'securityLight', 'desc':'Security Light', 'buttonPin': 17, 'relayPin':07},
		  'poolLight'    :{'code':'poolLight'    , 'desc':'Pool Light'    , 'buttonPin': 06, 'relayPin':12},
		  'gardenLight'  :{'code':'gardenLight'  , 'desc':'Garden Light'  , 'buttonPin': 13, 'relayPin':16},
		  'patioFountain':{'code':'patioFountain', 'desc':'Patio Fountain', 'buttonPin': 27, 'relayPin':20},
		  'frontFountain':{'code':'frontFountain', 'desc':'Front Fountain', 'buttonPin': 22, 'relayPin':21}};

var aDevices = [];

var deviceCounter = 0;

function oDevice (deviceName)
{
	var uniqueID = deviceCounter;
	this.id = deviceCounter;
	this.name = deviceName;
	this.code = deviceList[deviceName].code;
	this.desc = deviceList[deviceName].desc;
	this.buttonPin = deviceList[deviceName].buttonPin;
	this.relayPin = deviceList[deviceName].relayPin;
	this.status = true;
	this.active = true;

	this.button = new Gpio (this.buttonPin, 'in', 'rising', {debounceTimeout:50});

	this.relay = new Gpio (this.relayPin, 'out');
	this.relay.writeSync (1);

	this.button.watch (function (e,v) {watchProcess(e,v, uniqueID);});

	deviceCounter ++;
	console.log ("added an instance");
}

oDevice.prototype.flipRelay = function ()
{
	console.log ("Flipping the Relay : " + this.desc);
	this.status = ! this.status;

	var myValue = 0

	if (this.status == true) {myValue = 1;}
	this.relay.writeSync (myValue);
}

function watchProcess (e,v, id)
{
	
	console.log ("id: " + id);
	console.log ("Watch process changed the relay : " + aDevices[id].desc);
	aDevices[id].flipRelay();
}

for (var device in deviceList) 
{
	aDevices.push (new oDevice (device));
}

function count2(seconds)
{
	var waitTill = new Date(new Date().getTime() + seconds * 1000);
	while(waitTill > new Date()){}
	console.log("end of wait");
}

setInterval (function () {aDevices[0].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[1].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[2].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[3].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[4].flipRelay();}, 10000);

console.log ("Switch app started");
`
```

fivdi, 2017-06-06:
At the bottom of the code there is the following:
```js
setInterval (function () {aDevices[0].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[1].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[2].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[3].flipRelay();}, 10000);
//count2 (1);
setInterval (function () {aDevices[4].flipRelay();}, 10000);
```
This code toggles the state of each relay every 10 seconds. Please remove this code and try again.

Vercellotti, 2017-06-07:
Thank you for your response.
The 5 interval timers have been set on purpose to mimic the turning on and off of the circuits that will occur via timers that have been set in another part of the app. 

The strange behaviour I am getting is that the watchProcess callback is being called without an external trigger - i.e. no momentary switch was pressed or even connected to my RPi, so there should be no reason for the interupt to be called. 

Thoughts?

fivdi, 2017-06-07:
If nothing is connected to any of the input GPIOs and `watchProcess` is being called resulting in the following message being displayed
```
Watch process changed the relay : <aDevices[id].desc>
```
then maybe the inputs are floating. Normally each GPIO on the raspberry Pi has either a pull-up or a pull-down resistor enabled which means that it shouldn't float. Is there by any change other software that's being used and that's disabling the pull-up/pull-down resistors on the inputs?

Maybe the 8 channel relay is causing the problems. If the 8 channel relay is disconnected is `watchProcess` still called when it shouldn't be?

Does the simple example in the [usage section of the readme](https://github.com/fivdi/onoff#usage) work correcty for you?

fivdi, 2017-06-14:
@Vercellotti any news on this?

fivdi, 2017-06-18:
I'm going to assume that this was a hardware issue. If it turns out that it's a problem with onoff the issue can be reopened.

closed 2017-06-18 by fivdi