fivdi/onoff #73: Do I have to hold value in a loop?

c0debreaker, 2017-06-09:
I wrote this code and I see my relay module turns on like a split seconds and dies quickly. So looks like relay modules doesn't work like a light switch. We have to tell in continuously with a value of 1. Am I right?

```
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');
relay.writeSync(1);
```

c0debreaker, 2017-06-09:
Looking at your examples, you are using setTimeout. Isn't it bad for the relay if we are executing writeSync(1) infinitely?

I was wondering why in the command linux in linux, all I had to do was execute and the relay was in ON state until I sent a value of 0.
```
echo 1 > /sys/class/gpio/gpio4/value
```

fivdi, 2017-06-09:
Do you mean that the program turns the relay on and the program terminates immediately?
If so, this is to be expected with the following code:
```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');
relay.writeSync(1);
```
This program simply sets the state of GPIO4 to 1 and terminates because it has nothing else to do.

It should not be necessary to execute some sort of loop that continuously sets the value of GPIO4 to 1.

c0debreaker, 2017-06-10:
Got it. So yesterday I was reverse engineering my garage door opening. In order for it to open, I need to short the wire for 1 second. Using your `onoff` library, how can I set the value to 1 or HIGH for 1 second?

fivdi, 2017-06-10:
```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');

relay.writeSync(1);
setTimeout(function() {
  relay.writeSync(0);
}, 1000);
```

c0debreaker, 2017-06-10:
Thanks.

Do you have a method that can mimic what `echo 1 > /sys/class/gpio/gpio4/value` does? This commands sets the gpio to HIGH continuously without putting the code in a loop.

fivdi, 2017-06-10:
This should do it:
```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');

relay.writeSync(1);

// The rest of the application is here, for example
setInterval(function() {
  console.log(new Date());
}, 1000);
```

c0debreaker, 2017-06-10:
Sweet! I'll try it. Thanks a lot. Closing this ticket now. :)

c0debreaker, 2017-06-10:
The code below turns ON the relay module for 1 second then turns off. The led also turns on for 1 second then turns off. That's correct right?

```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');

relay.writeSync(0);
setTimeout(function() {
  relay.writeSync(1);
}, 1000);
```

fivdi, 2017-06-10:
>> The code below turns ON the relay module for 1 second then turns off.

What turns off?

>> The led also turns on for 1 second then turns off.

What LED?

c0debreaker, 2017-06-10:
I'm sorry about the confusion. There is a built-in LED on the relay module. When I set GPIO pin to HIGH, the LED also lights up.

https://www.amazon.com/SunFounder-Channel-Optocoupler-Expansion-Raspberry/dp/B00E0NTPP4/ref=sr_1_7?ie=UTF8&qid=1497064540&sr=8-7&keywords=relay+module+raspberry+pi

Using the code below, the built-in LED lights up and I hear a click. After 10 seconds, the LED light turns off and I also hear a click.

```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');

setTimeout(function() {
  relay.writeSync(1);
}, 10000);
```

c0debreaker, 2017-06-10:
This is weird, I changed the code to the below and the relay clicks and LED light turns ON. Why?

```js
var Gpio = require('onoff').Gpio;
var relay = new Gpio(4, 'out');

setTimeout(function() {
  relay.writeSync(0);
}, 10000);
```

fivdi, 2017-06-10:
This is from the product description at amazon:

> The 2 relays on the module are independent of each other, once a relay is triggered by the low level, the corresponding LED lights will be lit and accompanied by a crisp sound.

The behavior you are seeing is correct and expected.

Please read the documentation in the README. Everything is explained there. For example, in the documentation for the [Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) the `direction` argument is documented as follows

> direction - A string specifying whether the GPIO should be configured as an input or output. The valid values are: 'in', 'out', 'high', and 'low'. If 'out' is specified the GPIO will be configured as an output and the value of the GPIO will be set to 0. 'high' and 'low' are variants of 'out' that configure the GPIO as an output with an initial level of 1 or 0 respectively.

c0debreaker, 2017-06-10:
Thanks a lot!

c0debreaker, 2017-06-10:
Something is still not right. I used my voltmeter to make sure the labels on the relay module for NO and NC are correct. They are correct. Right now, the wire of the lamp light goes into COM and NO. I did a continuity test on COM and NO, there was no connectivity. That's good.

However, when I start my program, `node server.js`, the lamp light would turn ON. I haven't even called the /garageoff or /garageon yet.

To turn OFF the light, I would have to make an http call to http://localhost/api/garageon. And to turn it back ON, I would make an http call to api/garageoff. Weird right?

I think this code `var relay = new Gpio(4, 'out');`  sets GPIO pin 4 to high. I've tried it many times but it does really turn ON during start of the server.js

Here is my code
http://https://gist.github.com/c0debreaker/1dce444d29bf4ef0a6c9e2a93bdfc01c

fivdi, 2017-06-10:
> I think this code `var relay = new Gpio(4, 'out');` sets GPIO pin 4 to high

This is incorrect. PLEASE read the documentation as suggested above. This code sets GPIO4 (which is pin 7 on the GPIO expansion header) to low. The relay you are using is triggered by the low level. This code will therefore turn the relay on.

If you do not want to the relay to turn on when creating the Gpio object use the following code instead:
```
var relay = new Gpio(4, 'high');
```

c0debreaker, 2017-06-10:
Ah! And you're right about pin 7. I meant there GPIO4

c0debreaker, 2017-06-10:
This code didn't turn on the lamp during `node server.js` anymore. That's great!
```js
var relay = new Gpio(4, 'high');
```

I'll change my api endpoint to this
```js
router.route('/garageoff')

    .get(function(req, res) {

        var garage = req.body.garage;

        setTimeout(function() {
          relay.writeSync(1);
          res.json({ message: 'Garage door button switch was pressed.' });
        }, 1000);

    });

router.route('/garageon')

    .get(function(req, res) {

        var garage = req.body.garage;

        setTimeout(function() {
          relay.writeSync(0);
          res.json({ message: 'Garage door button switch was pressed.' });
        }, 1000);

    });
```

fivdi, 2017-06-10:
Why is `setTimeout` being called?

The call to `setTimeout` doesn't appear to serve any purpose other than delaying the calls to `relay.writeSync` and `res.json`  for no apparent reason.

c0debreaker, 2017-06-10:
I need to keep the wire shorted for at least 1 second before releasing the continuity. It's a requirement of the garage door opener to trigger it. If the wire was shorted for only less than a second, the garage door opener will not respond.

c0debreaker, 2017-06-10:
I actually wrote another endpoint last night specfically for opening of the garage door. I need to change the code above and rename it to lampon and lampoff. I forgot to correct the endpoint name.

Here is the code that I'm using now for the garage door. It's working great. It's the code you wrote earlier

```js
router.route('/garageshort')

    .get(function(req, res) {

        var garage = req.body.garage;

              relay.writeSync(0);
              setTimeout(function() {
              relay.writeSync(1);
              res.json({ message: 'Garage door button switch was pressed.' });
              }, 1000);

    });
```

c0debreaker, 2017-06-10:
Here is my latest code and it's working great.

http://192.168.1.76:8080/api/lamp/1 turns off relay
http://192.168.1.76:8080/api/lamp/0 turns on relay

https://gist.github.com/c0debreaker/67a6048c5023561efc96523326756b0a

fivdi, 2017-06-11:
Good to hear that everything is working now.

closed 2017-06-11 by fivdi