fivdi/onoff #77: Init issue

neophob, 2017-08-11:
Hi

Thanks for you GPIO library. I experienced a strange error, that recently happened. when I run this code in my application

```
const foo = new GpioOnOff(194, 'in').readSync();
const bar = new GpioOnOff(195, 'in').readSync();
```

this call fails with the error `ENOENT: no such file or directory, open '/sys/class/gpio/gpio194/edge'`.

When I start my application, none of the GPIO ports are exported. After the first (failed) start of my application, GPIO pin 194 is now exported. After a restart of my application, it will fail with the error `ENOENT: no such file or directory, open '/sys/class/gpio/gpio195/edge'` and GPIO pin 195 is exported.
And of course on the 3rd time I start my application it works as desired. 

I figured out, when I remove the init hack of this commit: https://github.com/fivdi/onoff/commit/7beb74b85485636b3772619e5d4bd0fd8dc12ae2#diff-107e8525cfb4da75b60dee3c069f4790 it works again.

When I read a gpio port, I guess there's no need that the `edge` file is present, right?

Ideas to resolve this issue:
- add option to disable the init hack
- the gpio init code is more clever and selects the needed gpio files depending on the configuration

What do you suggest?

fivdi, 2017-08-12:
@neophob Sorry about the modification that results in this issue. Making the initialization code more clever is the better idea.

What type of board were you using when you got the error?

mkatter, 2017-08-12:
Hi,
I got the same issue here on a **Nano Pi Neo Air**!
My App uses the GPIO 203, 64, 67 - all as direction out.
1st start: `Error: ENOENT: no such file or directory, open '/sys/class/gpio/gpio64/edge'` but it got exported.
2nd start: `Error: ENOENT: no such file or directory, open '/sys/class/gpio/gpio67/edge'` but it got exported.
3rd start: YAY! 👍 

203 has no issue as it supports hardware interrupts:
```
usr@sys:path# echo 203 > /sys/class/gpio/export
usr@sys:path# ls /sys/class/gpio/gpio203/
active_low  device  direction  edge  power  subsystem  uevent  value
```

Thanks for your work and time!

cu Markus

mkatter, 2017-08-12:
A quick 'n dirty hack for your init hack. It works for me but ... :)
```
    // A hack to avoid the issue described here:
    // https://github.com/raspberrypi/linux/issues/553
    // I don't like this solution, but it enables compatibility with older
    // versions of onoff, i.e., the Gpio constructor was and still is
    // synchronous.
    let paths = 
    [this.gpioPath + 'direction',
    //  this.gpioPath + 'edge',
     this.gpioPath + 'active_low',
     this.gpioPath + 'value',
    ]
    if (edge !== undefined) paths.push(this.gpioPath + 'edge')
    paths.forEach(function (path) {
      var tries = 0,
```

but, TBH, I absolutely agree 

> Making the initialization code more clever is the better idea.

cu Markus

neophob, 2017-08-12:
I use a custom board and like @mkatter pointed out, I read some (static) GPIO pins which do not support interrupts - thats where the init hack fails:

```
# ls /sys/class/gpio/gpio194/
active_low  device  direction  power  subsystem  uevent  value
```

fivdi, 2017-08-12:
Thanks for the feedback. I'll take a look at fixing this asap. That being said I won't be near a development computer before the end of August. One again apologizes for breaking things here.

fivdi, 2017-08-25:
Fixed with 80809fde8f5529feb0280c627bcdada95bc79539 and published on npm with onoff@1.1.7.

closed 2017-08-25 by fivdi

projetoarduino, 2017-12-12:
Hi i have same problem
ENOENT: no such file or directory, open '/sys/class/gpio/gpio67/edge'
My hardware is nano pi using 3.4.113 kernel

fivdi, 2017-12-12:
@projetoarduino It looks like GPIO67 is being used as an interrupt generating input and that there is code to create a Gpio object that looks something like this:
```js
var button = new Gpio(67, 'in', 'both');
```
Is this correct?

If this is correct, then I'd guess that the error is occurring because GPIO67 corresponds to GPIOC3 but the GPIOC port on the Allwinner H3 processor doesn't support interrupts. If my understanding of the datasheet is correct, on the Allwinner H3 only the GPIOA, GPIOG and GPIOL ports support interrupts. See table 3-2 starting on page 74 of the [Allwinner H3 Datasheet](http://wiki.friendlyarm.com/wiki/images/4/4b/Allwinner_H3_Datasheet_V1.2.pdf).

If what I'm saying here isn't correct please create a new issue to describe the problem and include code that can be used to reproduce the issue.

projetoarduino, 2017-12-13:
Hi @fivdi Thanks to reply
I use this code

var Gpio = require('onoff').Gpio,
in1 = new Gpio(203, 'in', 'none'),
in2 = new Gpio(6, 'in', 'none'),
in3 = new Gpio(64, 'in', 'none'),
in4 = new Gpio(65, 'in', 'none'),
in5 = new Gpio(66, 'in', 'none'),
in6 = new Gpio(67, 'in', 'none'),
Led = new Gpio(198, 'out');

fivdi, 2017-12-13:
@projetoarduino 

Please try with the following JavaScript code instead and let us know if it works. The 'none' parameter has been removed. To the best of my knowledge it should work and everything should function correctly thereafter.

```js
var Gpio = require('onoff').Gpio,
in1 = new Gpio(203, 'in'),
in2 = new Gpio(6, 'in'),
in3 = new Gpio(64, 'in'),
in4 = new Gpio(65, 'in'),
in5 = new Gpio(66, 'in'),
in6 = new Gpio(67, 'in'),
Led = new Gpio(198, 'out');
```

projetoarduino, 2017-12-13:
@fivdi 
I modified the code as you said, it works without problems, thanks for the help and congratulations for the work