fivdi/onoff #78: Push only button event

zafar-saleem, 2017-08-12:
I want to turn the LED on by pressing the button and then when I want to turn the LED off then I press the button again. That means I do not want to handle the release state. The LED will be on unless I press the button again to turn it off. Currently when I use the "watch" method then it listens to both press and release. The LED is on when I press the button and is off when I release the button. Is there a way to achieve this in onoff?

fivdi, 2017-08-12:
The `value` argument passed to the watch callback can be used to determine if the button was pressed or released. If `value` is 1 the button was pressed. If `value` is 0 the button was released. So if you're only interested in handling button press events then check to see if `value` is 1 in the watch callback. If it is, the button was pressed.

zafar-saleem, 2017-08-12:
Great I achieved it using a legacy(but still amazing) logic :).

`let counter = 0;

button.watch((error, value) => {
    if (value === 0 && counter >= 1) {
        led.writeSync(0);
        counter--;
        return;
    }

    if (value === 0 && counter <= 0) {
        led.writeSync(1);
        counter++;
        return;
    }
});`

fivdi, 2017-08-12:
Ok, good to hear.

closed 2017-08-12 by fivdi