fivdi/onoff #82: write or write sync won't work

TonyFeestneus, 2017-10-26:
Case:
GPIO.7 set as OUT (by gpio mode out)
gpio write 7 1 -> works
gpio write 7 0 -> works

But, this won't work:
```
var Gpio = require('onoff').Gpio,
  led = new Gpio(7, 'out');

led.writeSync(1);

process.on('SIGINT', function () {
  led.unexport();
});
```

How can this be? Does not give any errors... but doing 'gpio readall' shows no output change.

fivdi, 2017-10-26:
It looks like a Raspberry Pi is being used. Is this correct? If so, the gpio utility is a WiringPi utility and uses pin numbers specific to Wiring Pi. It doesn't use the standard Linux pin numbers (aka BCM GPIO pin numbers.) See [here](http://wiringpi.com/pins/) for more details.

The Linux pin numbers used by onoff on the Raspberry Pi can be seen [here](https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png). For example, pin 7 on the GPIO header is GPIO4 so the number to use with onoff is 4.

Try with the following to see if it works:
```js
var Gpio = require('onoff').Gpio,
  led = new Gpio(4, 'out');

led.writeSync(1);

process.on('SIGINT', function () {
  led.unexport();
});
```

TonyFeestneus, 2017-10-27:
Thanks, that works.

This has me very confused...

Your referenced pin number file: [https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png](url)
It names physical pin 7 as GPIO 4, while command 'gpio readall' lists physical pin 7 as GPIO 7.

Which is also confirmed by this pin diagram: [http://pi4j.com/images/j8header-2b-large.png](url)

I always get confused by this. What is what now?

fivdi, 2017-10-27:
@TonyFeestneus thanks for the feedback.

> Which is also confirmed by [this pin diagram](http://pi4j.com/images/j8header-2b-large.png)

Please take a look the text in the yellow box at the bottom of this image. It says that the numbers are WiringPi numbers, not Broadcom GPIO numbers. onoff does not use WiringPi numbers.

> I always get confused by this. What is what now?

As mentioned above onoff uses Broadcom GPIO numbers which can be seen [here](https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png).

I'm going to go ahead and close this issue as there is no actionable for onoff.

closed 2017-10-27 by fivdi