fivdi/onoff #85: readSync() doesn't work

jbeynar, 2018-02-05:
Consider following snippet:

`var onoff = require("onoff");
var Gpio = require('onoff').Gpio;

function toggleGpioPin(pinNumber){
    var gpioPin = new Gpio(pinNumber, "out");
    if (gpioPin.readSync() === 0) {
        gpioPin.writeSync(1);
        return 1;
    } else {
        gpioPin.writeSync(0);
        return 0;
    }
}

toggleGpioPin(7);
toggleGpioPin(7);`

I have found that readSync() doesn't really return actual value - it always  returns 0 instead 1 at the second toggleGpioPin(7) call.

I'm using Raspberry Pi A with Raspbian.

I've also used simplified code to ensure that readSync is a victime. Even if I have set mode to "in" it doesn't return proper value.

What am I doing wrong?

fivdi, 2018-02-05:
The first line of `toggleGpioPin` is
```js
var gpioPin = new Gpio(pinNumber, "out");
```
This instantiates a new Gpio object, configures the corresponding GPIO pin to be an output pin and sets the value of that pin to 0. It does this every time `toggleGpioPin`. This explains why `gpioPin.readSync()` always returns 0. To resolve the issue instantiate one Gpio object during the initialization phase of the program and always use that Gpio object. Something like this should work:
```js
var Gpio = require('onoff').Gpio;
var gpioPin = new Gpio(7, "out");

function toggleGpioPin(gpioPin){
    if (gpioPin.readSync() === 0) {
        gpioPin.writeSync(1);
        return 1;
    } else {
        gpioPin.writeSync(0);
        return 0;
    }
}

toggleGpioPin(gpioPin);
toggleGpioPin(gpioPin);
```

fivdi, 2018-02-05:
The documentation for the constructor should be improved to make it clear what it actually does.

fivdi, 2018-02-05:
The [documentation for the Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) already mentions that the GPIO is configured when a Gpio object is instantiated. With commit 0118ab9c1fae07c15c65c8b76700a58ce8f72f4d the documentation was tweaked to makes things clearer.

closed 2018-02-05 by fivdi