fivdi/onoff #86: How does one set the board mode using this module?

ThorstenBux, 2018-02-06:
Hi, 

I'd like to have control over the board mode pecify either Raspberry Pi or SoC/BCM pin schemas, by passing MODE_RPI or MODE_BCM. 

is this possible?

fivdi, 2018-02-06:
There are several different ways of doing pin numbering on the Raspberry Pi each with their advantages and disadvantages. What onoff uses are the GPIO numbers used by Linux to identify GPIOs. This is equivalent to MODE_BCM. This is also the only mode that onoff supports. Note that onoff runs not only on the Raspberry Pi but also on other computers like the BeagleBone Black, PocketBeagle and C.H.I.P. All of these computers support Linux GPIO numbers and this explains why these numbers are used by onoff.

Note that numbers don't play as big a role in onoff as in other libraries. The only place where numbers are used is when a new Gpio object is instantiated. It's not necessary to specify pin numbers when methods like read, write, watch, unexport etc. are called.

With a pin mapping it should be possible to achieve what you would like to achieve. For example, the [sample code from the usage section of the readme](https://github.com/fivdi/onoff#usage) could be implemented as follows if MODE_RPI pin numbering was needed:

```js
var Gpio = require('onoff').Gpio;

const PIN7 = 4;
const PIN11 = 17;

var led = new Gpio(PIN11, 'out');
var button = new Gpio(PIN7, 'in', 'both');

button.watch(function(err, value) {
  led.writeSync(value);
});
```

fivdi, 2018-02-07:
I think the questions from the OP has been answered so I'll go ahead and close this issue.

closed 2018-02-07 by fivdi