fivdi/onoff #87: I/O errors on first launch

chingizhumbatov, 2018-02-06:
Hi. I had some problems when I first launch as sudo this code snippet
```javascript
var Gpio = require('onoff').Gpio,
  led = new Gpio(12, 'out', 'both'),
  iv;

iv = setInterval(function () {
  led.writeSync(led.readSync() ^ 1);
}, 100);

setTimeout(function () {
  clearInterval(iv);

  led.writeSync(0);
  led.unexport();

  console.log('ok - ' + __filename);
}, 2000);
```

I got this error:

```
fs.js:540
  return binding.write(fd, buffer, offset, length, position);
                 ^
Error: EIO, i/o error
    at Object.fs.writeSync (fs.js:540:18)
    at Object.fs.writeFileSync (fs.js:987:21)
    at new Gpio (/home/pi/test/node_modules/onoff/onoff.js:122:10)

```
Subsequent launches are fine, until the next reboot.
After I changed this code snippet in onoff.js 
``` javascript
fs.writeFileSync(this.gpioPath + 'direction', direction);

if (edge) {
   s.writeFileSync(this.gpioPath + 'edge', edge);
}

if (!!options.activeLow) {
    fs.writeFileSync(this.gpioPath + 'active_low', ONE);
}
``` 
to 

```javascript
    try {
      fs.writeFileSync(this.gpioPath + 'direction', direction);
    } catch (ignore) {

    }

    try {
      if (edge) {
        fs.writeFileSync(this.gpioPath + 'edge', edge);
      }
      try {
        if (!!options.activeLow) {
          fs.writeFileSync(this.gpioPath + 'active_low', ONE);
        }
      } catch (ignore) {
      }
    } catch (ignore) {
    }

```

This error has been fixed. Is it possible for you to make these changes. thank you in advance

fivdi, 2018-02-06:
Is the issue occurring on a Raspberry Pi while logged on as user pi?
If so, is there a reason to use sudo? onoff should function without sudo.

chingizhumbatov, 2018-02-06:
I deleted the standard Pİ account and added my account. And when I run the above code snippet on behalf of the new user, it gives an error:
```
fs.js:663
  return binding.open(pathModule.toNamespacedPath(path),
                 ^

Error: EACCES: permission denied, open '/sys/class/gpio/export'

```

fivdi, 2018-02-06:
As far as I remember if the new user is added to group gpio it should work without  sudo. Can you give this a try please?

I don't think it's a good idea to make the change suggested above as the changes simply ignore the exceptions.

chingizhumbatov, 2018-02-06:
I added a new user to the GPİO group and  trying. But I got the same error which as sudo
```
fs.js:540
  return binding.write(fd, buffer, offset, length, position);
                 ^
Error: EIO, i/o error
    at Object.fs.writeSync (fs.js:540:18)
    at Object.fs.writeFileSync (fs.js:987:21)
    at new Gpio (/home/pi/test/node_modules/onoff/onoff.js:122:10)
```

fivdi, 2018-02-06:
I would imagine that the user isn't configured correctly to use gpio and that adding the user to group gpio isn't enough.

chingizhumbatov, 2018-02-06:
Thanks, I will try to reconfigure everything

fivdi, 2018-02-06:
If you flash a new SD card please check to see if it works for user pi before deleting user pi.

fivdi, 2018-02-10:
@chingizhumbatov was it possible to make any progress here?

chingizhumbatov, 2018-02-10:
I used a new SD card and tried to launch on behalf of the user pi, but I got the same error.

Hardware: Rpi 3
Raspbian version: Jessie
Node.js  version: v9.2.0

fivdi, 2018-02-10:
Ok, I now see what the problem is.

The following line of code from the first post:
```js
led = new Gpio(12, 'out', 'both'),
```
should be:
```js
led = new Gpio(12, 'out'),
```
Can you make this change, give it a try, and report your findings please?

The `edge` parameter `'both'` should only be specified for pins that support interrupts. `'out'` pins on the Raspberry Pi don't support interrupts, only `'in'` pins do. The [documentation for the edge parameter for the Gpio constructor](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) does mention this but it's not particularly easy to see.

I'm sorry for not spotting this earlier and even more sorry for getting you to flash a new SD card.

To be honest, I think this is bug. Old versions of Linux used to support interrupts on outputs as well as inputs. On new versions of Linux this is no longer the case. `onoff` should either ignore the `edge` parameter if it's specified for outputs or throw an error if it's specified.

fivdi, 2018-02-11:
@chingizhumbatov onoff@1.2.0 was published on npm today with a fix for this issue. When a Gpio object is instantiated for an output GPIO the edge argument passed to the Gpio constructor will be ignored. The documentation in the readme was also updated to make things a little clearer hopefully.

closed 2018-02-14 by fivdi