fivdi/onoff #89: `unexport` does not retain pin level of 0

RevealedFrom, 2018-03-08:
If I call `unexport`, there is some voltage at a pin even though I written a `0` to it.  It seems `unexport` will reset the pin to some default value that was there before `onoff` was imported.

What does `unexport` do? Is it critical to call it?

fivdi, 2018-03-08:
There's no mention of what type of computer is being used so I'll assume it's a Raspberry Pi.

The terms export and unexport are from the [Linux GPIO Sysfs Interface for Userspace Documentation](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt). onoff uses the GPIO sysfs interface to access GPIOs.

When a new onoff Gpio object is created what it does under the covers is to export the GPIO to userspace allowing the GPIO to be accessed by a "normal" user program from userspace. Calling the `unexport` method will reverses the effect of exporting to userspace. This is also all that is mentioned in the [Linux GPIO Sysfs Interface for Userspace Documentation](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt).

When a Raspberry Pi is booted, all GPIOs are configured as inputs and will have either the value 0 or 1. This can be verified by running the command `gpio readall` from the command line.

Your program created a Gpio object, configured it as an output and set the value to 0. This can also be verified by calling `gpio readall` from the command line while your program is running.

As soon as your program calls unexport, the pin will be configured as an input just like it was at boot time. The value 0 that your program wrote to the pin while it was configures as an output is no longer relevant as the pin is now an input. As usual, this can be verified with `gpio readall`. The value of the pin will probably be the same value that it was at boot time.

Calling `unexport` isn't critical as it will configure the pin as an input. The boot process also configures the pin as an input.

RevealedFrom, 2018-03-08:
Yes, it's a Raspberry Pi.  Thanks for the detailed explanation.

I was testing with a LED without a resistor. On booting up, the LED is lit dimly. At some point after I invoked `onfoff`, the LED went dark. After `unexport`, it became dimly lit again.

closed 2018-03-08 by RevealedFrom

fivdi, 2018-03-08:
I'd use a resistor.