fivdi/onoff #90: Error bindings.js

paulofreitasnobrega, 2018-03-26:
Good afternoon. I'm encountering the following error:

`/home/pi/Desktop/Test/node_modules/bindings/bindings.js:96
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/Desktop/Test/node_modules/epoll/build/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/build/Debug/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/build/Release/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/out/Debug/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/Debug/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/out/Release/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/Release/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/build/default/epoll.node
 → /home/pi/Desktop/Test/node_modules/epoll/compiled/8.9.4/linux/arm/epoll.node
    at bindings (/home/pi/Desktop/Test/node_modules/bindings/bindings.js:93:9)
    at Object.<anonymous> (/home/pi/Desktop/Test/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/Test/node_modules/onoff/onoff.js:4:11)
`
Test.js code:
`var Gpio = require('onoff').Gpio,
  led = new Gpio(17, 'out'),
  button = new Gpio(4, 'in', 'both');

button.watch(function(err, value) {
  led.writeSync(value);
});
`
Versions:
`pi@raspberrypi:~/Desktop/Test $ node --version
v8.9.4
pi@raspberrypi:~/Desktop/Test $ npm --version
5.8.0
`
Folder Structure:
`pi@raspberrypi:~/Desktop/Test $ ls
node_modules  package.json  test.js
pi@raspberrypi:~/Desktop/Test $ cd node_modules/
pi@raspberrypi:~/Desktop/Test/node_modules $ ls
bindings  epoll  nan  onoff
`
Any idea of the problem?

fivdi, 2018-03-26:
Good evening.

What was the exact command that was used to install onoff?
What was the complete output of that command?
Was Node.js v8.9.4 used to install onoff?

fivdi, 2018-03-26:
Does the file epoll.node exist in any of the locations listed in the error message?

paulofreitasnobrega, 2018-03-26:
@fivdi 
1. What was the exact command that was used to install onoff?
`sudo npm i onoff --save`

2. What was the complete output of that command?
At this moment I'm at work. I send the codes as soon as I get home.

3. Was Node.js v8.9.4 used to install onoff?
Yes

fivdi, 2018-03-26:
Why is sudo being used to install? User pi shouldn't need to use sudo to install in directories owned by pi themselves. The semantics of using sudo with npm are not that easy to understand, at least for me :smile: For example, when installing native modules that need to be compiled during the install process the --unsafe-perm option must be specified if sudo is used. epoll is a native module that onoff depends on. If it's really necessary to install with sudo the --unsafe-perm option should therefore also be specified. It would however be better not to use sudo as it shouldn't be needed here.

paulofreitasnobrega, 2018-03-26:
😄 Thank you @fivdi 

I'm crawling with linux 👶. However, if I'm not mistaken, I'd run a test without using `sudo`. Anyway, I redo the tests and make the results available soon.

paulofreitasnobrega, 2018-03-26:
I will redo the whole process:

```
pi@raspberrypi:~ $ node --version && npm --version
v8.9.4
5.8.0
```
```
pi@raspberrypi:~ $ cd Desktop/
pi@raspberrypi:~/Desktop $ sudo mkdir MyTest && cd MyTest
```

In this case I need to use sudo. Otherwise, I get an access error:

```
pi@raspberrypi:~/Desktop/MyTest $ sudo npm init -y
Wrote to /home/pi/Desktop/MyTest/package.json:

{
  "name": "MyTest",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "keywords": [],
  "author": "",
  "license": "ISC"
}
```
Without using sudo [Error]:
```
pi@raspberrypi:~/Desktop/MyTest $ npm i onoff --save
npm WARN checkPermissions Missing write access to /home/pi/Desktop/MyTest
npm WARN MyTest@1.0.0 No description
npm WARN MyTest@1.0.0 No repository field.

npm ERR! path /home/pi/Desktop/MyTest
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/home/pi/Desktop/MyTest'
npm ERR!  { Error: EACCES: permission denied, access '/home/pi/Desktop/MyTest'
npm ERR!   stack: 'Error: EACCES: permission denied, access \'/home/pi/Desktop/MyTest\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/home/pi/Desktop/MyTest' }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-03-26T20_50_32_803Z-debug.log
```
Using sudo and --unsafe-perm:
```
pi@raspberrypi:~/Desktop/MyTest $ sudo npm i onoff --save --unsafe-perm

> epoll@2.0.0 install /home/pi/Desktop/MyTest/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/MyTest/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/Desktop/MyTest/node_modules/epoll/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN MyTest@1.0.0 No description
npm WARN MyTest@1.0.0 No repository field.

+ onoff@2.0.0
added 4 packages from 9 contributors in 83.147s
```
Creating the test file:
```
pi@raspberrypi:~/Desktop/MyTest $ sudo touch test.js
pi@raspberrypi:~/Desktop/MyTest $ sudo nano test.js
```
Code:
```
var Gpio = require('onoff').Gpio,
  led = new Gpio(17, 'out'),
  button = new Gpio(4, 'in', 'both');

button.watch(function(err, value) {
  led.writeSync(value);
});
```
### Run - I have a different error now that I used the --unsafe-perm:
```
pi@raspberrypi:~/Desktop/MyTest $ node test.js
fs.js:735
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EBUSY: resource busy or locked, write
    at Object.fs.writeSync (fs.js:735:20)
    at Object.fs.writeFileSync (fs.js:1301:24)
    at new Gpio (/home/pi/Desktop/MyTest/node_modules/onoff/onoff.js:82:8)
    at Object.<anonymous> (/home/pi/Desktop/MyTest/test.js:3:12)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Function.Module.runMain (module.js:684:10)
```
In node_modules:
```
pi@raspberrypi:~/Desktop/MyTest/node_modules $ ls
bindings  epoll  nan  onoff
```
Any ideas on the new error?

fivdi, 2018-03-26:
Using `sudo mkdir MyTest` is a mistake and is the cause of the EACCES errors later on. Calling `sudo mkdir MyTest` creates a directory owned by root and user pi can not write to this directory without using `sudo`. As a general rule never use `sudo` unless it's essential and it's not essential when user pi wants to create a directory in directory `Desktop`.

Don't do this:
```
pi@raspberrypi:~ $ cd Desktop/
pi@raspberrypi:~/Desktop $ sudo mkdir MyTest && cd MyTest
```
Do this instead, i.e., with sudo:
```
pi@raspberrypi:~ $ cd Desktop/
pi@raspberrypi:~/Desktop $ mkdir MyTest && cd MyTest
```
It should then be possible to call `npm install onoff` without issues.

The EBUSY error is occurring because GPIO4 (for the button) is reserved for usage by something else. Because it's reserved by something else your application can't use it. That something else is most likely the 1-wire interface. Did you use the raspi-config tool to enable 1-wire? If so, this would explain the EBUSY error as the 1-wire interface reserves GPIO4. Please use the raspi-config to disable 1-wire if it's enabled and try again. It may be necessary to reboot after disabling 1-wire.

fivdi, 2018-03-26:
Please create new comments when adding additional information rather than editing old comments to add the information. Your last comment above changes each time I look at it which makes it very difficult to follow.

fivdi, 2018-03-26:
After commenting out `dtoverlay=w1-gpio` do a reboot.

paulofreitasnobrega, 2018-03-26:
@fivdi After restarting, it worked ... 😄
Thank you for the patience. One last question. If I comment the lines:

```
# dtoverlay = dwc2
# dtoverlay = w1-gpio
```

**What is the effect of this change on my raspberry?**

Thanks a lot for the help.

fivdi, 2018-03-27:
For information related to `dwc2` see [here](https://gist.github.com/gbaman/50b6cca61dd1c3f88f41).
For information related to `w1-gpio` see [here](https://pinout.xyz/pinout/1_wire#).

Note that neither of these options are enabled by default on Raspbian so it looks like you enabled them yourself. If you don't need them the disabling them should not result in issues.

closed 2018-03-27 by fivdi

paulofreitasnobrega, 2018-03-27:
Thanks a lot for the help @fivdi