fivdi/onoff #91: Timmer in Dimmer

paulofreitasnobrega, 2018-04-13:
Good afternoon! Sorry for the mild question that is probably being posted in an incorrect location. But I do not know where to publish.

I am using my raspberry pi to control the power of a 60W bulb in an AC network, 127V, 60Hz. To help raspberry in this task, I've developed a PCB that has a TRIAC BTA41600 and some other components. This PCB synchronizes with the network and manipulates the sine wave, cutting it and obtaining different voltages at the end.

I have run several tests and the PCB seems to work perfectly. My problem is with the javascript code.

I am using, in my raspberry, the nodejs and the onoff module. I make a time routine using setTimeout ().

Before the code, I'd like to quickly explain how it should work:
My network frequency is 60Hz, so each sine wave has a period of 16.66 milliseconds. Each wave has two cycles (positive / negative). In this way, each cycle has a period of 8.33 ... milliseconds. To size this wave, I need to divide it into the number of desired graduations, in my case I used 100 (which would represent 100%). Therefore, for each wave graduation I have a period of 0.083 milliseconds. To summarize: When I set a 30% graduation, I receive a period of 2.49 milliseconds. This period is the time that my TRIAC will wait, within the cycle, to be activated.

Follow the code:

```
//onoff
const gpio = require('onoff').Gpio;
const zc = new gpio(17, 'in', 'falling');
const dimmer = new gpio(27, 'out') //shot triac

//time
const frequency = 60
const drivingTRIAC = 8.33 * 0.001
const wave = (((1 / frequency) * 1000) / 2)
const resolution = (wave - drivingTRIAC) / 100

//power for time
var power = 30
var powertime = resolution * (100 - power)

//zero crossing
zc.watch(function (err, value) {    
    setTimeout(() => {
        // ON TRIAC
        dimmer.writeSync(1)
        setTimeout(() => {
            // OFF TRIAC
            dimmer.writeSync(0)
        }, drivingTRIAC)
    }, powertime)
});
```

When turning this code, my lamp receives a non-linear flashing effect. However, my timing system is working normally. When doing a test with the code below, I get as a result: 120 (with some variations of 119 or 121):

```
const gpio = require('onoff').Gpio;
const zc = new gpio(17, 'in', 'falling'); //zero crossing
const pulse = 0;

zc.watch(function (err, value) {
    pulse += 1;
    console.log(pulse);
});
```
I guess my problem is related to the misuse of setTimeout or some function. can you help me?

fivdi, 2018-04-14:
I would imagine that `setTimeout` isn't accurate enough for what you would like to achieve. If I'm not mistaken you would like to use setTimeout to wait multiples of 0.083 milliseconds, i.e., n * 0.083 where n is an integer. setTimeout is not suitable for this type of precision. setTimeout can be used to wait n milliseconds where n is an integer and depending on the system load it may not be very accurate.

closed 2018-04-14 by fivdi

paulofreitasnobrega, 2018-04-14:
I did not know that information. Would you know any way to do this with nodejs? Is there any way to accurately calculate time?

paulofreitasnobrega, 2018-04-14:
I did a test with Date.now (), comparing the time of passing through zero (zc.watch), with the activation moment of the triac (8.33 milliseconds later). But I got the same effect on my lamp. Probably the code was not developed correctly, because in the zc.watch call I put a while (status) loop, where status got a true by default and a false when it reached the time of 8.33 milliseconds. I'll redo this test code and post it here so you can review it.

paulofreitasnobrega, 2018-04-14:
I did the test using the while and conditional. I also used console.time to mark execution times. One of the problems I realized was that my raspberry is taking a lot longer to respond to events. (powertime = exact waiting time to trigger triac; A: time the raspberry waited to trigger the triac; B: time the raspberry waited to turn off the triac) See below:

```
powertime: 5.8309999999999995 ms
A: 14.576ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 7.274ms
B: 0.192ms
powertime: 5.8309999999999995 ms
A: 6.586ms
B: 0.137ms
powertime: 5.8309999999999995 ms
A: 13.772ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 18.417ms
B: 0.132ms
powertime: 5.8309999999999995 ms
A: 12.178ms
B: 0.139ms
powertime: 5.8309999999999995 ms
A: 6.529ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 16.261ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 9.824ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 17.661ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 7.158ms
B: 0.136ms
powertime: 5.8309999999999995 ms
A: 15.598ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 9.601ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 12.785ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 14.697ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 13.019ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 1.960ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 9.367ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 6.062ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 13.021ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 10.938ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 12.956ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 14.467ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 10.066ms
B: 0.141ms
powertime: 5.8309999999999995 ms
A: 14.501ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 13.560ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 10.923ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 9.996ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 15.937ms
B: 0.136ms
powertime: 5.8309999999999995 ms
A: 10.794ms
B: 0.136ms
powertime: 5.8309999999999995 ms
A: 11.156ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 11.918ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 6.417ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 9.108ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 11.210ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 6.368ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 9.568ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 10.096ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 8.575ms
B: 0.137ms
powertime: 5.8309999999999995 ms
A: 2.718ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 14.159ms
B: 0.132ms
powertime: 5.8309999999999995 ms
A: 9.275ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 11.691ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 11.902ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 8.185ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 11.089ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 13.329ms
B: 0.133ms
powertime: 5.8309999999999995 ms
A: 15.366ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 10.914ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 9.905ms
B: 0.134ms
powertime: 5.8309999999999995 ms
A: 6.599ms
B: 0.211ms
powertime: 5.8309999999999995 ms
A: 6.762ms
B: 0.135ms
powertime: 5.8309999999999995 ms
A: 26.157ms
B: 0.135ms
^Cpowertime: 5.8309999999999995 ms
A: 9.263ms
B: 0.137ms
powertime: 5.8309999999999995 ms
A: 10.256ms
B: 0.134ms
```
below the code:

```
// pins
const gpio = require('onoff').Gpio
const zc = new gpio(17, 'in', 'falling')
const dimmer = new gpio(27, 'out')

// wave senoidal and cycles
// ((1 / 60) / 2) * 1000 = 8.33ms
const cycle = 8.33
const graduation = cycle / 100

// power = 30
const powertime = graduation * (100 - 30)

// Functions timer
const createFutureTime = function (milliseconds) {
    return new Date(Date.now() + milliseconds)
}

const isFutureTime = function (futureTime) {
    return Date.now() >= futureTime.getTime() ? true : false
}

//loop
var startTime = false
var futureTime = false
var triac = false

var myTest = function () {
    while (startTime) {
        if (!futureTime) {
            futureTime = createFutureTime(powertime)
            console.time('A')
        }

        if (isFutureTime(futureTime)) {
            if (!triac) {
                // console.log('triac on')
                console.log(`powertime: ${powertime} ms`)
                dimmer.writeSync(1)
                triac = true
                console.timeEnd('A')
                console.time('B')
                futureTime = createFutureTime(0.1)
            } else {
                // console.log('triac off')
                console.timeEnd('B')
                dimmer.writeSync(0)
                triac = false
                futureTime = false
                startTime = false
            }
        }
    }
}

// zero crossing
zc.watch(function (err, value) {
    startTime = true
    myTest()
})

// quit
process.on('SIGINT', function () {
    zc.unexport();
});
```

I ask again. Do you know of any way to keep my raspberry more synchronized?
Thank you for all the help you have already given me.

fivdi, 2018-04-14:
I'm not aware of a reliable way to achieve what you would like to achieve.

paulofreitasnobrega, 2018-04-14:
Good afternoon. I found the problem! It is related to synchronization and probably to the 10k pull-up resistor. I made a test code to check the number of pulses per cycle (In my case, at 60Hz, it should be 120). This test pointed out that I am getting false-positive at various times. Watching the lamp, it is precisely in these moments that it oscillates.

I looked at the README of the onoff module, and realized that there are guidelines for using pull-up and pull-down. But I did not quite understand how to use it. Could you guide me? (I'm aware, I've asked for a lot of help for you).

Follow the test code. In it pin 17, which receives the synchronizing pulses, is configuring as the descent edge:

```
// pins
const gpio = require('onoff').Gpio
**const zc = new gpio(17, 'in', 'falling')**
const dimmer = new gpio(27, 'out')

var us = require('microseconds')
const graduation = 83
const powertime = graduation * (100 - 80)
var loop = false
var triac = false
var timeStartCycle = false
var pulses = 0

var cycle = function () {
    while (loop) {
        if (us.since(timeStartCycle) >= powertime) {
            if (!triac) {
                dimmer.writeSync(1)
                triac = us.now()
            } else if (us.since(triac) >= 8.33) {
                dimmer.writeSync(0)
                triac = false
                timeStartCycle = false
                loop = false
            }
        }
    }
}

// zero crossing
**zc.watch**(function (err, value) {
    **if (value == 1) {
        console.log(value)
    }**
    timeStartCycle = us.now()
    loop = true
    cycle()
    pulses++
})

setInterval(() => {
    console.log(pulses)
    pulses = 0
}, 1000)

// quit
process.on('SIGINT', function () {
    zc.unexport();
});
```

The results of my raspberry pi.  (At the moment I receive 120 pulses stable, my lamp also stabilizes):

```
1
1
1
1
1
1
99
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
119
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
**120
121
120
120
121
120
120
121
120
120
121**
1
1
1
1
1
1
1
1
1
1
1
1
118
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
121
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
117
1
1
1
121
```

fivdi, 2018-04-14:
Is the question how to configure pullup / pull down resistors on the Raspberry Pi?

paulofreitasnobrega, 2018-04-15:
Yes. :(

fivdi, 2018-04-15:
And what were the problems?

How will using the internal ~50k pullup resistor rather than the external 10k pullup resistor that you are currently using change things?

paulofreitasnobrega, 2018-04-15:
I received the same results. This is really discouraging me. :(

I've made new tests that might indicate something to someone more experienced than me.

1st Test: I decide to do the tests from the beginning, in stages. In this way, I set up a synchronization circuit, using the same components, in a protoboard. (Power control components such as TRIAC or MOC3020 optocoupler were not included in this test). In nodejs code, only a descent edge tracking and a count of pulses. Result: 120 pulses as expected.

2nd Test: In this, I used the same code as the first test. But this time, the test was performed on the PCB that I had assembled (it is ready, with all components welded). Result: 120 pulses.

3rd Test: At that moment, I connected my raspberry to the PCB pin that controls TRIAC firing. However, I did not change the code, I kept the same of the last tests. The hope in this test was to find some anomaly realigned to the firing pin, but nothing abnormal appeared. Result: 120 pulses.

4th Test: Here the scenario has changed. By including a code to control TRIAC firing, my synchronization has stopped working. Result: 80 to 120 pulses, alternately. I performed tests using setTimeout and also While (true) with time control through Date.now () and third-party modules: microseconds and nanotimer. In all cases the result was the same.

Please, analyze the result below. It represents 1 second of circuit operation. Notice the trailing edge # 2. Note that the cycle () function was called, but the setTimeout () of this cycle was simply ignored. Another point that you can observe is the final amount of cycles in 1 second: 91, in this case.

If math is applied: 1/60 (grid frequency) = 16 ms per complete cycle. Each complete cycle has two half cycles. That is: 8.3 ms per half-cycle. With 91 cycles = 91 * 8.3 = Approximately 755 ms. A loss of approximately 245 ms per second. (In a perfect scenario this should be: 120 * 8.3 = Approximately 1000 ms - no rounding).

```
0: off triac
=============
1: set shot triac
1: on triac
1: off triac
=============
2: set shot triac
2: off triac
=============
3: set shot triac
3: on triac
3: off triac
=============
4: set shot triac
4: on triac
4: off triac
=============
5: set shot triac
5: on triac
5: off triac
=============
6: set shot triac
6: on triac
6: off triac
=============
7: set shot triac
7: on triac
7: off triac
=============
8: set shot triac
8: on triac
8: off triac
=============
9: set shot triac
9: on triac
9: off triac
=============
10: set shot triac
10: on triac
10: off triac
=============
11: set shot triac
11: on triac
11: off triac
=============
12: set shot triac
12: on triac
12: off triac
=============
13: set shot triac
13: on triac
13: off triac
=============
[... 348 lines cut ...]
});

/**
 * Define um loop que se repetirá a cada 1 segundo (1000ms). Ao final de cada
 * período, imprime o valor final de pulses e o zera
 */
iv = setInterval(function () {
    console.log(`pulses: ${pulses}`);
    pulses = 0;
}, 1000);

/**
 * Quando o script é interropido (ctrl + C), limpa os processos
 */
process.on('SIGINT', function () {
    clearInterval(iv);
    zc.unexport();
    dimmer.writeSync(0);
});
```
When removing the call line the cycle () function (within zc.watch ()). Notice how the number of pulses increases substantially.

`pulses: 115`

### **So follow my question. Do you believe that raspberry is unable to work with such a low time accuracy?**

paulofreitasnobrega, 2018-04-15:
This C code is working perfectly on my Arduino Uno and also on my NodeMCU v1. Realize that it's basically the same thing. Anyway, I'm unhappy that I can not replicate this on nodejs.

```
#define loadR D2 
volatile int power = 100;  
 
void zero_crosss_int()  
{
  // 60Hz-> 8.33ms (1/2 ciclo)
  // (8333us - 8.33us) / 256 = 32
  int powertime = (32*(256-power));      
  delayMicroseconds(powertime);   
  digitalWrite(loadR, HIGH);  
  delayMicroseconds(8.33);  
  digitalWrite(loadR, LOW);   
}
 
void setup()
{
  Serial.begin(9600);
  pinMode(loadR, OUTPUT);  
  attachInterrupt(D4, zero_crosss_int, RISING);  
}
 
void loop()
{
  power=10;
  delay(1000);
  power=60;
  delay(1000);
  power=120;
  delay(1000);
  power=180;
  delay(1000);
  power=240;
  delay(1000);
}
```