fivdi/onoff #92: export Gpio availability

towc, 2018-04-15:
I'm using this package on a raspberry pi, but sometimes I want to run some of the code on my laptop to test it out, and I'd rather not have to tell the code that this machine can't use GPIOs through some config file, because it's easily detectable.

I'd like to be able to do `if (Gpio.available) { actual code } else { mock code }`,  so I can change my code based on that. For one of the projects, I used try/catch, but it's inelegant and unneeded, and now I'm facing the same problem on another project.

towc, 2018-04-15:
fixed in PR https://github.com/fivdi/onoff/pull/93

towc, 2018-04-15:
well, it works. What's difficult about it?

fivdi, 2018-04-15:
I doesn't look like it works to me. See the review.

towc, 2018-04-15:
Did you submit it? I'm not seeing any review :/
I have to admit I don't have an rpi near me, so I just assumed this would give a true positive on it, but guess not

fivdi, 2018-04-15:
Sorry, I forgot to hit the submit button. To me it looks like the PR wasn't tested.

towc, 2018-04-15:
ok, I tried to hack together a quick solution. Again, I don't have an rpi around atm to test it out :/ And couldn't find the contribution guides.
What I did in another project, was having a module that tries to create a `new Gpio`, and if that errored out, then that module told the rest of my modules to use some mock code. It's not elegant, but it should cover the huge majority of cases on all operating systems. Can you take care of it?

johntalton, 2018-04-21:
@towc given the current PR could you validate (on your laptop) that this function will now return false as appropriate.

Tested on RPI, RPI + Docker, with and w/o root priv.

also, laptop specs?  arch? os? i2c hw? etc

fivdi, 2018-05-13:
@towc @johntalton Thank you both for your work on the pull request for this feature :+1:

onoff@3.1.0 which has just been published on npm now has a new static property called [accessible](https://github.com/fivdi/onoff#static-accessible) which can be used to determine whether or not GPIO access is possible.

closed 2018-05-13 by fivdi