fivdi/onoff #94: configuring the pin changes the pin's state

bluetiger30, 2018-04-16:
Hi , 
I noticed that when I configure the pin to act as an output :

`var pin= new Gpio(17,'out')`

it will automatically set the pin to low, is it possible to disable this? as I don't want it to set the value of that pin to any value when the programme reboots for some reason.

fivdi, 2018-04-16:
If onoff is used to configure a pin as an output it's not possible to say it shouldn't be given a value. With onoff a pin can only be configured as an output with an initial value of 0 or 1.

To configure the pin as an output with an initial value of 0 use either:
```js
var pin= new Gpio(17,'out')
```
or:
```js
var pin= new Gpio(17,'low')
```
To configure the pin as an output with an initial value of 1 use:
```js
var pin= new Gpio(17,'high')
```
See [here](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options).

To be honest, I don't know how onoff could know whether a program is starting normally for the first time or abnormally for a second time because the program reboots (restarts) for some reason.

bluetiger30, 2018-04-17:
Hi fivdi , Thanks for your response.

Yes I had a look at the high and low options however they don't resolve it as I want it to be initiated to what the pin is set to already.

just to give you an example why this is an issue for me: I'm using the module to control a relay.the script will check the time and based on the current time it will turn the pin high or low.

If I had the pin on high and the pi reboots and the gpio is defined as low and the time on the script wats it to be high then the relay will flick between high then low then high instantaneously which will damage the appliance connected to the relay.

I'm thinking of storing the pin status on a file and checking it once the script runs, or using on off to read the pin and then set it as an output 

I'm happy to hear if you have anyother suggestions

bluetiger30, 2018-04-18:
Jut wanted to keep you updated @fivdi :
stating null did the trick when the programme restarts, however, it throws errors if the pi reboots :
1st run 
```
fs.js:735
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EINVAL: invalid argument, write
    at Object.fs.writeSync (fs.js:735:20)
    at Object.fs.writeFileSync (fs.js:1301:24)
    at new Gpio (/home/pi/Projects/Moda/node_modules/onoff/onoff.js:65:10)
    at Object.<anonymous> (/home/pi/Projects/main.js:11:16)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Function.Module.runMain (module.js:684:10)
```
2nd run
```
fs.js:735
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EPERM: operation not permitted, write
    at Object.fs.writeSync (fs.js:735:20)
    at Gpio.writeSync (/home/pi/Projects/Moda/node_modules/onoff/onoff.js:174:8)
    at /home/pi/Projects/Moda/main.js:407:18
    at Job.job (/home/pi/Projects/Moda/main.js:347:17)
    at Job.invoke (/home/pi/Projects/Moda/node_modules/node-schedule/lib/schedule.js:173:10)
    at /home/pi/Projects/Moda/node_modules/node-schedule/lib/schedule.js:543:11
    at Timeout._onTimeout (/home/pi/Projects/Moda/node_modules/node-schedule/lib/schedule.js:501:7)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
```
For my application , I'm going to store the pin stuts and set the pin configration based on the last saved status.

Adding something similar to your library would be beneficial for future uses 

kind regards

fivdi, 2018-04-18:
@bluetiger30 Thanks for the update.

> For my application , I'm going to store the pin stuts and set the pin configration based on the last saved status.

This seam like the most reasonable solution to me.

> Adding something similar to your library would be beneficial for future uses

I'm not sure I agree here. I don't think onoff should decide how to configure pins and what their values should be. I think this is something that the application should do.

closed 2018-04-19 by bluetiger30