fivdi/onoff #97: Using GPIO as the stepper pulse

chadlinden, 2018-04-21:
Hi,  

I'm not sure if I'm over-thinking this, but I want to drive a stepper motor. Here is a Gist of my test code, https://gist.github.com/chadlinden/2a9c43c0e533433b90229420e2ee8634  

When I used a less granular timer, e.g. `new Date().getTime()` everything worked smoothly, but it was too slow. So I use node's process' hrtime to have send pulses faster. What's happening now is that I'm getting the stepper up to speed, there seems to be 'ticks' where pulses are being queued, or blocked.  

At the peak (shortest interval between pulses) it's sending a pulse every 0.000367 seconds. Is there a better method to handle a stepper motor? If not, is this a limitation of the library or Pi 3b+?

fivdi, 2018-04-21:
It looks like there's a problem [here](https://gist.github.com/chadlinden/2a9c43c0e533433b90229420e2ee8634#file-test-js-L17):
```js
 var nano = parseInt(process.hrtime().join(''));
```
and the same problem [here](https://gist.github.com/chadlinden/2a9c43c0e533433b90229420e2ee8634#file-test-js-L26):
```js
 var newNano = parseInt(process.hrtime().join(''));
```
Does it work better if they are changed to this:
```js
  var hrtime = process.hrtime();
  var nano = hrtime[0] * 1e9 + hrtime[1];
```
and this:
```js
  var newhrtime = process.hrtime();
  var newNano = newhrtime[0] * 1e9 + newhrtime[1];
```

> Is there a better method to handle a stepper motor?

There's a better way to do everything :). The code in the gist is blocking synchronous code so it will block everything in the Node process while running. You'll need to decide whether this is ok for the use case in question. If it's not ok then maybe it would be better to use `setInterval` to step every few milliseconds. This however may mean that the motor will not run fast enough for the use case.

fivdi, 2018-04-27:
There doesn't appear to be any todos for onoff here so I'll go ahead ans close the issue.

closed 2018-04-27 by fivdi