fivdi/onoff #98: event multiple occurences

msartor92, 2018-05-05:
Hi, I'm using this library on a pi to watch on input event. When i test it manually connecting the input pi to 3.3V pin the event occur once till i detach the connection as I expect.

Insted if I connect the input pin to 3.3V pin through a relay, even if the relay state is stable, I'm getting multiple event fires.
The library fires a rising event at each slot of debounceTime that i set at 2 sec.
When the input pin will move to 0 volt I'm getting a falling event at each debounceTime slot.

There is a way to debug this case? Any other that get same use case? It's a bug?

Thanks

msartor92, 2018-05-06:
[image]

Maybe with the picture is simply to understand. The goal is get a signal from the relay to generate an event. Below there is an abstraction of the code:

let button = new Gpio(17, 'in', 'rising', {debounceTimeout: 2000});
button.watch(function(err, value) { console.log('event'); });

The circuit and the software works quite well. The only problem I'm getting is when the realy switch state from 87a to 87 I'm reciving the event code side. But I'm getting lots of event (one each 2 sec) not only one. To be more precise the relay swtich state and maintain it.
I hope that now is more clear.

fivdi, 2018-05-09:
Please post a complete program that can be used to reproduce the problem.

fivdi, 2018-05-13:
@msartor92 The relay is not being used the way it should normally be used in the circuit diagram above.

It looks like it's an automotive relay. If this is the case then pin 30 is the common pin, pin 87 is the normally open contact and pin 87a is the normally closed contact. While the circuit diagram looks like it should work it assumes that their is no additional circuitry inside the relay that would prevent it from working.

Perhaps it would work better if pin 30 was used as the common pin (connected to 3.3v) and pin 87 (normally open contact) was connected to GPIO17. Pin 87 would also need to be connect to ground via a strong pull-down resistor. When the relay is triggered there would then be a rising edge on pin 87.

fivdi, 2018-05-25:
I'm going to go ahead and close this issue as it doesn't look like it's an onoff issue. If it turns out that it is an onoff bug the issue can be reopened.

closed 2018-05-25 by fivdi