fivdi/onoff #101: Runnning on Raspberry pi 3b+ Getting this error

Rainsster, 2018-05-12:
/home/pi/BotFiles/RaspberryPiProject/node_modules/bindings/bindings.js:88
        throw e
        ^

Error: /home/pi/BotFiles/RaspberryPiProject/node_modules/epoll/build/Release/epoll.node: invalid ELF header
    at Object.Module._extensions..node (module.js:598:18)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at bindings (/home/pi/BotFiles/RaspberryPiProject/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/pi/BotFiles/RaspberryPiProject/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)

fivdi, 2018-05-12:
`onoff` depends on a native C++ module called [epoll](https://github.com/fivdi/epoll). When `onoff` is installed `epoll` is also installed. `epoll` is also compiled because it's a native module. It looks like the binary for `epoll` which is in file `/home/pi/BotFiles/RaspberryPiProject/node_modules/epoll/build/Release/epoll.node` was not created for the Raspberry Pi.

Is it possible that `onoff` was compiled on one computer which is not binary compatible with the Raspberry Pi and then copied to the Raspberry Pi?

fivdi, 2018-05-14:
@Rainsster were you able to make any progress here?

fivdi, 2018-05-17:
I'm going to go ahead and close this issue as there doesn't appear to be anything that needs to be changed in onoff.

Rainsster, 2018-05-18:
Thank you for the help, i reinstalled it and it worked perfectly.