fivdi/onoff #102: Switch on and off a lamp using relay

muscaiu, 2018-05-12:
Is there an example for switching a simple lamp on and off using a relay?
I'm using a raspbery pi and a 5v relay, i'm able to switch it on and off using python code, but i'm not into python. I would like to use node and js instead.

**This is my setup:**
[image]

PS: the W3Schools example for turning a led on and off worked fine, but when using a relay things get confusing.

Thank you!

fivdi, 2018-05-12:
There isn't an example in the repo and although it can be a little tricky normally relays should work with onoff.

The important thing about relays is that some models are triggered (turned on) by a high value while other models are triggered by a low value.

I'm guessing here but I think your relay is triggered by a low value. If a relay is triggered by a low value it's important to ensure that the Gpio output object used to control the relay is configured to have a high value when the Gpio object is created. This prevents the relay from turning on automatically when the Gpio object is created. In other words, create the Gpio object for the relay like this:
```js
const relay = new Gpio(21, 'high'); // IMPORTANT: Use 'high' if relay uses low level trigger
```
rather than like this:
```js
const relay = new Gpio(21, 'out');
```
Reboot your Raspberry and give the following program a try:
```js
const Gpio = require('onoff').Gpio;
const relay = new Gpio(21, 'high'); // IMPORTANT: Use 'high' if relay uses low level trigger

console.log('Program started. Relay should be off. Relay will turn on in 5 seconds.');

setTimeout(() => {
  relay.writeSync(0);
  console.log('Relay should be on. Relay will turn off in 5 seconds.');

  setTimeout(() => {
    relay.writeSync(1);
    console.log('Relay should be off. ');
  }, 5000);
}, 5000);
```
If this doesn't resolve the issue please explain what the issue is an post a short program that can be used to reproduce the issue.

See the [Gpio constructor documentation](https://github.com/fivdi/onoff#gpiogpio-direction--edge--options) for more details.

muscaiu, 2018-05-13:
Only **the first 2 lines of code will turn on the relay**.  Tried all options (high, low, out),
```
const Gpio = require('onoff').Gpio;
const relay = new Gpio(21, 'high'); //same thing with 'low' or 'out'
```
The only option that doesn't start it seems to be 'in'.
```
const Gpio = require('onoff').Gpio;
const relay = new Gpio(21, 'in');

console.log('Program started. Relay should be off. Relay will turn on in 2 seconds.');

setTimeout(() => {
  relay.writeSync(0);
  console.log('Relay should be on. Relay will turn off in 2 seconds.');

  setTimeout(() => {
    relay.writeSync(1);
    console.log('Relay should be off. ');
  }, 2000);
}, 2000);

process.on('SIGINT', function () {
  relay.unexport();
});

```
However, using the 'in' option gives me an error:
**OUTPUT:**
```
Program started. Relay should be off. Relay will turn on in 2 seconds.
fs.js:786
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EPERM: operation not permitted, write
    at Error (native)
    at Object.fs.writeSync (fs.js:786:20)
    at Gpio.writeSync (/home/pi/soft/bec/node_modules/onoff/onoff.js:174:8)
    at Timeout.setTimeout (/home/pi/soft/bec/git_example.js:7:9)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

```

fivdi, 2018-05-13:
>Only the first 2 lines of code will turn on the relay. Tried all options (high, low, out),

>const Gpio = require('onoff').Gpio;
>const relay = new Gpio(21, 'high'); //same thing with 'low' or 'out'

What you say here means that everything turns the relay on. Setting the value of GPIO21 to 0 turns the relay on and setting the value of GPIO21 to 1 turns the relay on. If this is really the case the relay does not work. 

>However, using the 'in' option gives me an error:

This is to be expected as it's not possible to call writeSync for a GPIO input. writeSync can only be called for GPIO outputs.

- Please post the Python program that can be successfully used to control the relay
- Please post a link to the documentation for the relay
- Can you ever hear the relay clicking on and off?
- When does the red LED on the relay turn on and off?
- When does the green LED on the relay turn on and off?
- Are you sure that the lamp is wired correctly to the relay?
- What type of lamp is being used?
- Please describe how the lamp is connected to the relay (a photo would be good here)

muscaiu, 2018-05-13:
Please post the Python program that can be successfully used to control the relay
```
import RPi.GPIO as GPIO
import time
channel = 21
# GPIO setup
GPIO.setmode(GPIO.BCM)
GPIO.setup(channel, GPIO.OUT)
def motor_on(pin):
    GPIO.output(pin, GPIO.HIGH)  # Turn motor on
def motor_off(pin):
    GPIO.output(pin, GPIO.LOW)  # Turn motor off
if __name__ == '__main__':
    try:
        motor_on(channel)
        time.sleep(2)
        motor_off(channel)
        time.sleep(2)
        GPIO.cleanup()
    except KeyboardInterrupt:
        GPIO.cleanup()
```

    Please post a link to the documentation for the relay
https://www.optimusdigital.ro/en/relay-modules/874-relay-module-ordered-with-5v.html

    Can you ever hear the relay clicking on and off?
Yes, it's clicking on and off when using the python code and when the light turns on, and also when using the onoff module 2 lines of code. However, when using the onoff code, the light and relay remains on, and i have to run the python code to turn the light and relay off.

    When does the red LED on the relay turn on and off?
The red light is always on.

    When does the green LED on the relay turn on and off?
The green light turns on when the relay clicks. 

    Are you sure that the lamp is wired correctly to the relay?
Well you can see the connected wires in the picture and it works with the python code, so yes.

    What type of lamp is being used?
It's just a lightbulb plugged into the house power source basically, i have attached 2 more pictures.

    Please describe how the lamp is connected to the relay (a photo would be good here)
[image]
[image]

fivdi, 2018-05-13:
@muscaiu Thank you for all the information.

I now have a relay connected up to the Raspberry Pi and tested it with the Python code and the following code:
```js
const Gpio = require('onoff').Gpio;
const relay = new Gpio(21, 'out');

function motor_on(gpio) {
  gpio.writeSync(1);
}

function motor_off(gpio) {
  gpio.writeSync(0);
}

motor_on(relay);
setTimeout(function () {
  motor_off(relay);
  setTimeout(function () {
    relay.unexport();
  }, 2000);
}, 2000);
```
On my Raspberry Pi both the Python code and the JavaScript code turn the relay on for two seconds and then turn it off. Can you give this JavaScript code a try with Node.js pelase?

If this code doesn't work on your Raspberry Pi can you give the following code a try please. Note that it uses GPIO6 which is pin 31 on the GPIO header so you will need to modify your circuit accordingly.
```js
const Gpio = require('onoff').Gpio;
const relay = new Gpio(6, 'out');

function motor_on(gpio) {
  gpio.writeSync(1);
}

function motor_off(gpio) {
  gpio.writeSync(0);
}

motor_on(relay);
setTimeout(function () {
  motor_off(relay);
  setTimeout(function () {
    relay.unexport();
  }, 2000);
}, 2000);
```
The difference between GPIO6 and GPIO21 is that the internal pullup resistor is activated on GPIO6 by default but on GPIO21 the internal pulldown resistor is activated by default. Perhaps this difference is the source of the issue.

muscaiu, 2018-05-13:
Hurray!!! The hard part is over! After 2 weeks of failed attempts! Thank you so much man! 

It works well now on either of the gpios (21 or 6). I switched to 6 now, since mentioned something about being better :-D

The next step is to read the pin status, so i can send it to a webpage that has a ON-OFF switch.
I will try to adapt it to the [W3Schools WebSocket tutorial](https://www.w3schools.com/nodejs/nodejs_raspberrypi_webserver_websocket.asp) code.

The final step should be just like in the [W3Schools WebSocket tutorial](https://www.w3schools.com/nodejs/nodejs_raspberrypi_webserver_websocket.asp), the ability to switch it on and off from phone, witch worked fine with a simple LED.

fivdi, 2018-05-13:
>Hurray!!! The hard part is over! After 2 weeks of failed attempts! Thank you so much man!

Congratulations :tada: and thank you :smile: 

Have fun with the rest of the tutorials and onoff...

fivdi, 2018-05-13:
That is the expected behaviour of the program that you just posted. Calling `relay.unexport()` will free the resources associated with GPIO21 and reconfigure it as an input GPIO before exiting the program. Reconfiguring GPIO21 as an input results in the relay being turned off.

fivdi, 2018-05-13:
Please describe exactly what you would like to achieve.

muscaiu, 2018-05-13:
I'm trying to say that `gpio.writeSync(1)` and `gpio.writeSync(0)` is doing nothing, 
That's why i posted the code without it so you can see that the behavior is the same, yet there is no control over the relay.

fivdi, 2018-05-13:
Let's take a different approach to see if the underlying sysfs files used by onoff can be used to achieve anything.

First, power down the Pi, pull out the power plug for the Pi and stick the plug back in to restart the Pi. After restarting and logging as user `pi` don't run any Python or JavaScript programs.

Do everything as user `pi`. Don't use `sudo` for anything.

Start a terminal window and execute the following commands one by one. Don't copy and paste all the commands at the same time. Run them one at a time waiting for each command to complete before starting the next command.

```
cd /sys/class/gpio
ls -l
echo 6 > export
cd gpio6
ls -l
echo out > direction
cat value
```
Running the command `cat value` should print 0, does it?

Now run the following command:
```
echo 1 > value
```
The state of the relay should have changed, did it?

Now run the following command:
```
echo 0 > value
```
The state of the relay should have changed again, did it?

Now run the following command:
```
echo 1 > value
```
The state of the relay should have changed again, did it?

There should be no errors while running any of the above commands. If there are errors something is wrong. It should be possible to change the state of the relay by echoing 0/1 to the file called value. If the state of the relay can't be changed there is also something wrong.

fivdi, 2018-05-13:
I missed the following question.
Please don't edit comments to add important information. Create a new comment.

> PS: i'm assuming that relay.unexport() is being used to release the resources and not to control the relay, correct?

Incorrect. Calling [unexport](https://github.com/fivdi/onoff#unexport) reverses the effect exporting. Calling unexport for an output GPIO will reconfigure it as an input. This is also highly likely to change the state of the relay.

muscaiu, 2018-05-14:
I followed you instructions and this is what's happening:
- `echo out > direction` : the relay is being switched on (gren light on) 
- `cat value` :  prints 0
- `echo 1 > value` : nothing is hapening
- `echo 0 > value` : nothing is hapening

What is your next suggestion?
Maybe change the wiring? 
Try another relay?
I have 2 more relays as seen in this [picture ](https://user-images.githubusercontent.com/11426309/39965007-a8892e48-5699-11e8-8e19-043a1a6be445.JPG)(1 ssd and 1 similar to the current except it has 2 chanels).

muscaiu, 2018-05-14:
I'm connecting via ssh from my laptop and using user pi and no sudo like you said.

fivdi, 2018-05-14:
Pull out the plug on the lamp and retry

fivdi, 2018-05-14:
If it's still not working disconnect the lamp from the relay and retry.

muscaiu, 2018-05-14:
Disconected the lamp completely from the relay and same thing is happening.
When i run `echo out > direction`  the relay is being switched on (gren light on)

[image]

fivdi, 2018-05-14:
>  When i run `echo out > direction` the relay is being switched on (gren light on)

This is to be expected. Before configuring the GPIO as an output by calling `echo out > direction` the GPIO is configured as an input. The GPIO is actually configured as an input as soon as the Raspberry Pi is powered on. The relay remains off as long as the GPIO is configured as an input. Configuring the GPIO as an output by calling `echo out > direction` also sets the value of the GPIO to low (0V). The relay appears to recognize this and turns the relay on. What it doesn't recognize is high (3.3V) levels. 

Do you have a multimeter to measure the voltage on the GPIO to see if it's high enough to be recognized as a high level when the value file is set to 1?

As you mention above:

Perhaps the relay is broken... You could try some of your other relays to see if they work.
It doesn't appear to be the wiring but you could change the jumper wires to see if it helps.

At this point we can say that this isn't an onoff issue.

fivdi, 2018-05-17:
I'm going to go ahead and close this issue as there doesn't appear to be anything that needs to be changed in onoff.

closed 2018-05-17 by fivdi

barry786-dev, 2022-04-19:
Hi, this kind of relay is triggered on a low level, so without details: just connect the VCC of the relay with only 3.3V supply of raspberry, now when you start your node js code, follow the code as fivdi mentioned above use ('high'), that will write value 1 and the relay will not be triggered, when you want to start your relay to connect use writeSync(0) and when you want to disconnect use writeSync(1)... if you want to understand why then you have to read more about how relays are working, here small article about that I hope it will help. https://electronics.stackexchange.com/questions/202680/relay-what-is-the-difference-between-low-level-trigger-and-high-level-trigger