fivdi/onoff #104: Why does the API use 1/0 and not true/false?

pizzaisdavid, 2018-05-19:
Thanks for the nice library y'all.

Some of the other gpio libraries support writing true and false as HIGH and LOW.
```
gpio.write(false) // works by happy accident
gpio.write(true) // does not work
``` 

I'm curious:
what decisions led to using 1/0 over true/false
 if a pull request that added support for true/false would be welcome

fivdi, 2018-05-19:
> Thanks for the nice library y'all.

You're welcome :smile:

> Some of the other gpio libraries support writing true and false as HIGH and LOW.

Which libraries? Some of them also use 0 and 1.

> what decisions led to using 1/0 over true/false

- Under the covers [onoff uses sysfs files](https://github.com/fivdi/onoff#how-does-onoff-work) located at /sys/class/gpio to access GPIOs. The values written to the various sysfs files are vales like "in", "out", "high", "low", "none", "both", "rising", "falling" and of course 0 and 1. The onoff API also uses all of these values. onoff doesn't attempt to hide or abstract these values. All of these values are all visible to onoff users. To set the value of a GPIO HIGH or LOW the value 1 or 0 is written to the corresponding value file, not the value true or false. People familiar with these sysfs files will expect to use the values 1 and 0 rather than true and false.
- When I think about binary numbers I think about a sequence of 0s and 1s. I don't think about a sequence of trues and falses. I see a GPIO as a 1 bit number that is 0 or 1. I don't see it as true or false.
- For me LOW is the same as 0V (Volts) so giving it the value 0 rather that false seems more logical.
- If LOW is 0 then HIGH is obviously 1 :smile:
- Personal preference.

> if a pull request that added support for true/false would be welcome

I don't think so. I think it will complicate things. If write and writeSync start accepting false and true in addition to 0 and 1 the question as to what read and readSync return then arises. readSync currently returns 0 or 1 and this can't change. However, if someone calls gpio.writeSnyc(true) and it functions as expected they are likely to expect gpio.readSync() to return true. This would not be the case.

I'd be more inclined to throwing an error in write and writeSync if the value passed was not 0 or 1.

pizzaisdavid, 2018-05-20:
> Which libraries?

- https://www.npmjs.com/package/rpi-gpio

Looking around more it seems using 1/0 are way more common.

fivdi, 2018-05-20:
> Looking around more it seems using 1/0 are way more common.

Yes, for example [Johnny-Five](https://github.com/rwaldron/johnny-five/blob/v0.15.0/eg/board-with-port.js#L17) also uses 1/0.

To avoid using constants directly in onoff code Gpio.LOW and Gpio.HIGH can be used.

closed 2018-05-20 by fivdi