fivdi/onoff #106: Excpetion on macOS

rgwch, 2018-05-21:
I'm developing on a Mac, so I tried the feature Gpio.accessible() to mock the GPIO on the Mac. But this is not possible since require()'ing onoff gives the following error:

```
Error: Module did not self-register.
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:707:18)
    at Module.load (internal/modules/cjs/loader.js:589:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:528:12)
    at Function.Module._load (internal/modules/cjs/loader.js:520:3)
    at Module.require (internal/modules/cjs/loader.js:626:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at bindings (/..basedir../node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/..basedir../node_modules/epoll/epoll.js:1:99)
    at Module._compile (internal/modules/cjs/loader.js:678:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:689:10)
```

So I have to stick with a manual "const realpi=false" solution and require('onoff').Gpio only on the Raspberry (where it works perfectly). 

* OS: macOSX High Sierra
* Node: 10.1.0
* onoff: 3.1.0

fivdi, 2018-05-21:
@rgwch Yes, you're right. `Gpio.accessible()` only works correctly on Linux. It will not work correctly on macOS or Windows.

fivdi, 2018-05-21:
This should be documented in the readme.

johntalton, 2018-05-22:
pull request #107 adds readme note. with the assumption this is attributed to eopoll linux stickiness. (aka __linux__ ifdef guard)

... and as such is not a `Gpio.accessible()` specific issue

rgwch, 2018-05-22:
Thanks for the quick response and the epoll explanation! It's not a big problem, of course, and onoff is still a great helper, but a few words in the documentation about this might indeed save quite a bit of time.

fivdi, 2018-06-13:
I'm going to go ahead and close this issue as a note has been added to the documentation explaining how onoff only functions correctly on Linux.

@rgwch Thanks for reporting the issue
@johntalton Thanks for create the PR

closed 2018-06-13 by fivdi