fivdi/onoff #109: Use with "open drain" interrupt pins?

GeoffreyPlitt, 2018-05-23:
I'm connecting an MCP23017's INTB pin to GPIO26 on a Pi Zero.

The MCP23017's documentation includes open-drain mode, which is highly recommended in forums, but how do I configure this library/GPIO to detect interrupts in that mode?

johntalton, 2018-05-23:
[rpio gpio](https://www.raspberrypi.org/documentation/usage/gpio/) documentation makes a short note about pull-up/down resistors.

also [Table 6-31](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) gives you the full default state for all (for rpi case the shorthand is anything about 8 is pull-low

In short, just prevent the line from floating and use `edge` / `activeLow` to negate how you interpret the interrupt.

SysFs doesn't allow control of the Broadcom pull-up/down state (as far as i know).  Though it can be configured via other methods.

johntalton, 2018-05-23:
Using the following code snip i was able to see activity from the Mcp23 on its INTA and INTB pins (wired to boradcom [5](https://pinout.xyz/pinout/pin29_gpio5) and [6](https://pinout.xyz/pinout/pin31_gpio6) pins respectively)

```
const onoff = require('onoff');

const pin5 = new onoff.Gpio(5, 'in', 'both', { activeLow: true });
const pin6 = new onoff.Gpio(6, 'in', 'both', { activeLow: true });

console.log('current pin 5 value', pin5.readSync());
console.log('current pin 6 value', pin6.readSync());

pin5.watch((err, value) => { console.log('watch pin 5 event', err, value); });
pin6.watch((err, value) => { console.log('watch pin 6 event', err, value); });
```

fivdi, 2018-05-23:
What @johntalton says should work if the INTB pin is connected to a GPIO where "Pull" is "High" in [Table 6-31](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf). 

@johntalton Why is the option `activeLow: true` specified? I don't think this is necessary.

It's also possible to use an external pull-up resistors. For example, the MCP9808 temperature sensor also has a open-drain interrupt pin called alert. It can be connected to a GPIO as shown [here](https://raw.githubusercontent.com/fivdi/mcp9808-temperature-sensor/master/examples/circuit.png).

johntalton, 2018-05-23:
If the pull-Up (HIGH) on pin 5 (our INTB pin connected to the mcp23 open-drain) keeps the pin set to HIGH (1); when there is an interrupt (open drain is closed), the line is pulled LOW (0):

`activeLow` flips the bit so that from code read will return HIGH as Interrupt condition.

... or, im completely thinking about it wrong :) (working code is not always a sign of workin conceptual model)

GeoffreyPlitt, 2018-05-23:
So helpful, thanks!

closed 2018-05-23 by GeoffreyPlitt