fivdi/onoff #110: "Bad file descriptor" error while unexport gpio

msalafia, 2018-06-01:
Hi, i have used the following code at the end of my script in order to free the resources:

```js
process.on('SIGINT', function () {
    util.freeResources();
});
```
where util is my external module with the following function definition:

```js
module.exports.freeResources = function() {
    LED.unexport();
}
```

When i stop my application with `CTRL + C` nothing happen and the application seems frozen. When i push `CTRL + C` a second time it returns the following:

```
fs.js:629
  return binding.close(fd);
                 ^

Error: EBADF: bad file descriptor, close
    at Object.fs.closeSync (fs.js:629:18)
    at Gpio.unexport (/home/pi/Desktop/DemoNodeUa/node_modules/onoff/onoff.js:241:8)
    at Object.module.exports.freeResources (/home/pi/Desktop/DemoNodeUa/utility.js:45:9)
    at process.<anonymous> (/home/pi/Desktop/DemoNodeUa/index.js:92:10)
    at process.emit (events.js:160:13)
    at Signal.wrap.onsignal (internal/process.js:198:44)
```

Am i doing something wrong?

fivdi, 2018-06-01:
Hi, can post a complete program that is as short as possible that can be used to reproduce the problem please?

Also, which versions of Node.js and on off are being used?

msalafia, 2018-06-01:
You can try the project cloning [this](https://github.com/msalafia/RaspUAServer) end doing `git checkout led`.

I use Node v9.4.0 and onoff 3.1.0.

fivdi, 2018-06-01:
I would imagine that the SIGINT handler is being called twice. Once each time ctrl+c is hit. The program doesn't terminate automatically the first time ctrl+c is hit because there are still asynchronous operations pending that have not completed yet somewhere in your code. The second time the SIGINT handler is called the program does terminate because an error is thrown due to the attempt to unexport the GPIO a second time. The easiest way to resolve this issue would be to call `process.exit(0)` at the end of the SIGINT handler.

msalafia, 2018-06-01:
Thank you @fivdi, this solved my issue!

closed 2018-06-01 by fivdi

ChocolateLoverRaj, 2022-09-04:
> I would imagine that the SIGINT handler is being called twice. Once each time ctrl+c is hit.

Calling `process.exit(0)` fixed the problem for me too. I am confused about why SIGINT is being called twice even though I only did ctrl + c once.

fivdi, 2022-09-04:
> > I would imagine that the SIGINT handler is being called twice. Once each time ctrl+c is hit.
> 
> Calling `process.exit(0)` fixed the problem for me too. I am confused about why SIGINT is being called twice even though I only did ctrl + c once.

You would need to debug the code to figure out why.