fivdi/onoff #111: Permission denied on Rock64 with debian

MarlonBe, 2018-06-06:
So I am trying to get the most basic thing ever to work:
```javascript
const Gpio = require('onoff').Gpio;
const led = new Gpio(17, 'out');
```
And I am getting:
```
fs.js:646
    return binding.open(....)....
Error: EACCESS: permission denied, open '/sys/class/gpio/export'
```

fivdi, 2018-06-06:
This isn't an `onoff` issue. It's an issue related to the how permissions have been setup for the user that ran the program and got the `EACCESS` error. The user doesn't have the permissions that are required to access files at `/sys/class/gpio` but these permissions are needed by `onoff` to use GPIOs.

There's more than one way to resolve this issue.

#### Alternative 1
Use `sudo`:
```
sudo node your-app-name-here.js
```

#### Alternative 2
I'm not familiar with the ROCK64 but it should be possible to setup udev rules which will enable the user to access the files at `/sys/class/gpio` without `sudo` privileges.

For example, on a Raspberry Pi, if the user is in group `gpio`, adding the following rules to `/etc/udev/rules.d/99-com.rules` will enable the user to access GPIOs without `sudo` privileges:
```
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
	chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
	chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
'"
```
Please note that the exact details of the udev rules that are needed on a ROCK64 are very likely to differ to those required on the Raspberry Pi.

fivdi, 2018-06-06:
The information at http://synfare.com/599N105E/hwdocs/rock64/index.html indicates that the ROCK64 doesn't have GPIO-17 available on a GPIO header but the JavaScript code posted above uses GPIO-17.

MarlonBe, 2018-06-06:
The rules sadly did not work and I don't want to run it with sudo privileges.
I'm using a Pi now and it works like a charm <3

closed 2018-06-06 by MarlonBe

dkebler, 2019-03-13:
This may be a kernel issue more than rock64

This single line worked for me in kernel 4.15 on an ubuntu 18.04 rpi distro
```
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'find -L /sys/class/gpio/ -maxdepth 2 -exec chown root:gpio {} \; -exec chmod 770 {} \; || true'"
```

which i found here
https://stackoverflow.com/a/50650398/4695378

I have some rock64pros I want to use on/off with so will report back here.
What I don't know is if the dto for gpios has been built into the kernel by default, but I assume it being important was.  
https://github.com/ayufan-rock64/linux-build